prusa3d/PrusaSlicer #1: C++14 issues

flannelhead, 2016-11-01:
Compiling from master @ https://github.com/prusa3d/Slic3r/commit/36f51ff2bfb03af88e2121703578817fca13f0d9 yields quite a many compilation errors with a recent `gcc` (version 6.2.1) on Linux due to errors in code in `#ifdef SLIC3R_CPP > 11` blocks. The default `C++` standard for recent `gcc` is `C++14` (more precisely `-std=gnu++11` which contains GNU extensions). The errors are especially related to rvalue reference overloaded functions in e.g. `PolylineCollection.cpp` (`chained_path` and `chained_path_from`).

Are you accepting contributions to your fork? I might start slowly looking at these in order to make it build on more recent compilers and take advantage of the move semantics.

flannelhead, 2016-11-01:
Also, I should mention that the code builds fine with the same compiler with the `-std=c++11` flag. So there are definitely faults in the `C++14` specific code.

bubnikv, 2016-11-01:
Thank you. I am developing mainly on Visual Studio 2013 and the builds are built on Strawberry Perl and on MacOS, where the C++14 features are disabled. I shall disable the C++14 optimizations for now, before the issues get cleared.

bubnikv, 2016-11-01:
> Are you accepting contributions to your fork? I might start slowly looking at these in order to make it build on more recent compilers and take advantage of the move semantics.

Yes, please, we will be thankful for pull requests.

flannelhead, 2016-11-01:
@bubnikv Alright, I shall create a PR where I incrementally add the fixes. At least I'll be able to fix the most evident mistakes, but the other stuff might take some time. I'll close this issue for now.

flannelhead, 2016-11-01:
Reopening to make this better discoverable by those having similar problems.

flannelhead, 2016-11-03:
@bubnikv I assume this can be closed now as https://github.com/prusa3d/Slic3r/pull/10 got merged.

closed 2016-11-03 by flannelhead