prusa3d/PrusaSlicer #2: 'only overhang layer has more than one speed' test failed

CReimer, 2016-11-01:
While compiling this for Arch Linux there's an error message while running through the tests.

```
t/perimeters.t ................. 46/59 
#   Failed test 'only overhang layer has more than one speed'
#   at t/perimeters.t line 281.
#          got: '0'
#     expected: '1'
t/perimeters.t ................. 59/59 # Looks like you failed 1 test of 59.
t/perimeters.t ................. Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/59 subtests 
```

bubnikv, 2016-11-02:
Thanks, but this is a little too little to draw any conclusion from.

What Arch Linux? What version? 32bit or 64bit? What gcc version? What perl version? It would greatly help, if you could provide a build log, which contains the compilation flags with which the GCC compiler is executed.

CReimer, 2016-11-02:
Yes, of course. There's no Arch Linux version it's a rolling release distribution.
It's 64bit
gcc version is 6.2.1 20160830
perl version is v5.24.0
[buildlog.txt](https://github.com/prusa3d/Slic3r/files/566230/buildlog.txt)

And here's the PKGBUILD script: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=slic3r-prusa3d
It's still a bit of a mess, but I'm yet to find an easier solution to compile slic3r into a distribution package.

flannelhead, 2016-11-02:
I can also confirm this. Same OS and software versions as @CReimer. I'm not building with the above `PKGBUILD` script but manually with locally installed Perl packages using [these instructions](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux).

flannelhead, 2016-11-03:
Reverting https://github.com/prusa3d/Slic3r/commit/c443f49da3e968841f43dc1f51aae189239c12d3 does seem to fix this test for me but fails another test "seam is aligned". @CReimer, can you confirm?

I might investigate this a bit further as there is something C++11 related (`std::swap`) going on in that commit.

flannelhead, 2016-11-03:
Well, seems to not be C++11 related. Compiling with `-std=c++98` (the standard that's default in older `gcc`) gives the same test failure.

bubnikv, 2016-11-03:
Looks like I would have to install the ArchLinux beauty into a virtual machine.

flannelhead, 2016-11-03:
@bubnikv I guess any Linux distro with a recent enough `gcc` (preferable >= 6.1) would be good. Arch Linux is probably a bit of a hassle to set up (you'll have to roll the install manually step by step).

Otherwise, it wouldn't be a bad idea to test the builds on a Linux VM. Fedora, maybe? Or Debian testing.

CReimer, 2016-11-03:
If installing Arch Linux is too much of a hassle you can always use Antergos instead.
Even though it's not as beautiful as Arch Linux :smile:

bubnikv, 2016-11-03:
We run our builds on Debian, but the server may not be quite up to date.

CReimer, 2016-11-03:
I'm not sure how helpful it is, but that's the content of %layer_speeds in perimeters.t

```
# $VAR1 = {
#           '0.350' => {
#                      '3960' => 1
#                    },
#           '4.250' => {
#                      '3960' => 1
#                    },
#           '3.050' => {
#                      '3960' => 1
#                    },
#           '1.850' => {
#                      '3960' => 1
#                    },
#           '1.550' => {
#                      '3960' => 1
#                    },
#           '4.550' => {
#                      '3960' => 1
#                    },
#           '3.350' => {
#                      '5940' => 1
#                    },
#           '5.450' => {
#                      '3960' => 1
#                    },
#           '0.950' => {
#                      '3960' => 1
#                    },
#           '4.850' => {
#                      '3960' => 1
#                    },
#           '2.150' => {
#                      '3960' => 1
#                    },
#           '6.050' => {
#                      '3960' => 1
#                    },
#           '2.450' => {
#                      '3960' => 1
#                    },
#           '6.350' => {
#                      '3960' => 1
#                    },
#           '5.150' => {
#                      '3960' => 1
#                    },
#           '5.750' => {
#                      '3960' => 1
#                    },
#           '3.950' => {
#                      '3960' => 1
#                    },
#           '2.750' => {
#                      '3960' => 1
#                    },
#           '1.250' => {
#                      '3960' => 1
#                    },
#           '3.650' => {
#                      '3960' => 1
#                    },
#           '0.650' => {
#                      '3960' => 1
#                    }
#         };
```

bubnikv, 2016-11-03:
> If installing Arch Linux is too much of a hassle you can always use
Antergos instead.
> Even though it's not as beautiful as Arch Linux 😄

There are about 5x more Linux distributions then there are Linux
programmers.

Running the Arch Linux installation. This reminds me of my early 1999
encounters with Slackware Linux, installling from a shoe box of floppies.

flannelhead, 2016-11-03:
@CReimer @bubnikv isn't that the expected result? I see one of the layers has a different speed 5940 while the others have 3960. There's something strange going on with the test itself.

CReimer, 2016-11-03:
On 03.11.2016 19:02, bubnikv wrote:
> Running the Arch Linux installation. This reminds me of my early 1999
> encounters with Slackware Linux, installling from a shoe box of floppies.
>

For reasons I don't understand they want to keep their setup like that. 
I would have switched to another distribution if Arch weren't that easy 
to customize.

bubnikv, 2016-11-03:
It was not that bad, at least into the virtualbox. I kinda like it, I think I understand why people like 'back to the roots' style.

Running the Build.PL. Let's see.

flannelhead, 2016-11-03:
@bubnikv Nice! I will do also some testing on Ubuntu 16.10 on my end to see if it's distro-specific or not.

flannelhead, 2016-11-03:
The same test seems to fail on Ubuntu 16.10 as well. `gcc 6.2.0` and `Perl 5.22.2` there.

bubnikv, 2016-11-03:
> Reverting c443f49 does seem to fix this test for me but fails another test "seam is aligned".

Good catch, @flannelhead . Thanks! I introduced this error in c443f49. While I tried to avoid copying a vector of points, I did not intialize the split paths correctly. Following checkin fixes this issue.

https://github.com/prusa3d/Slic3r/commit/483a658144a2a59cd0f7f7eea4bcac8cbb45eb44

closed 2016-11-03 by bubnikv

flannelhead, 2016-11-04:
@bubnikv Great to see it's fixed, thank you! 🎉 `git bisect` is sometimes a wonderful tool 😉