prusa3d/PrusaSlicer #3: Infill not touching perimeters on slopy walls

Sebastianv650, 2016-11-01:
Nice to see someone want's to give Slic3r a little boost!

### Version
Release 1.31.2

### Operating system type + version
Win 10 64bit

### Behavior
See the red marked areas in the screenshot attached, I used the cat from the thingiverse link at the end of the post. The infill is not touching the perimeter line. As long as the walls are nearly vertival (lower areas of the model) they are touching the perimeters. This is related to "Ensure vertical shell thickness": Turn it off and the problem is removed. See the blue marked screenshot, but also there in the circular area there is a small zig-zag area as it would "Ensure vertical shell thickness"?

Aditionaly, in the red marked one, there are tiny infill line segments in the circular area. I can find such "zits" in the most layers, original Slic3r isn't doing that?
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
http://www.thingiverse.com/thing:1703197

bubnikv, 2016-11-02:
Thanks for the report. The "ensure vertical wall thickness" inserts a full infill spot at the sloping surfaces. That's why you are seeing the zig-zags on the bottom image. The gaps between the perimeter and the infill are likely caused by the "ensure vertical wall thickness" infill spot being too small to fit the zig-zag infill into. This is an error and I hoped to fix it already. I will look into that.

Sebastianv650, 2016-11-02:
Yes, but only the top image where the perimeters are not touching Ensure vertical shell thicknessis enabled. The lower picture had it disabled, so i guess it's also a bug.

bubnikv, 2016-11-03:
I hope the gaps between the solid infill and the sparse infill were significantly reduced by
https://github.com/prusa3d/Slic3r/commit/61d82b02b527066803584bc3bcfd3da852caebab

It is difficult to fit two infill regions one on another exactly, without introducing gaps or overfills. I believe I achieved a reasonable compromise. Please reopen this issue if you find the solution insufficient.

closed 2016-11-03 by bubnikv