prusa3d/PrusaSlicer #4: Slicer just crash

Eriobis, 2016-11-02:
### Version
1.31.2

### Operating system type + version

Windows 7

### Behavior
When I open the slic3r, load a file then the screen is not responding.

I don't think a screenshot is necessary. The perl interpreter crash after sometimes

#### STL/Config (.ZIP) where problem occurs

Eriobis, 2016-11-02:
[WER61F3.tmp.appcompat.txt](https://github.com/prusa3d/Slic3r/files/565449/WER61F3.tmp.appcompat.txt)

bubnikv, 2016-11-02:
Thanks for the report.

What windows 7 version? 32bit? 64bit? what service pack? What Slic3r? 32bit or 64bit?

You did not manage to attach the STL/Config files. So it hangs, but does not crash per se? Is it specific to a single STL, or is it a general problem? Is it a large STL? How much RAM do you have?

It really helps, if you could give as much detail as possible.

Vojtech

Eriobis, 2016-11-02:
I'm using Windows 7 64 bits with the 64 bits slic3r on the latest windows 7 ( I suppose service pack 1 )
I have 32g of ram.
The software hang for a while, until I see the perl interpreter crash window. ( Can I change the version of perl or all the dependencies come with slic3r? )

I coudn't send the stl , even in .zip github wouldn't let me upload the file ( its a small file tho ). I will upload it elsewhere and send you the link

bubnikv, 2016-11-04:
Isn't it a duplicate of a following issue?
https://github.com/prusa3d/Slic3r/issues/11#issuecomment-258402600

Does the crash disappear if you go to preferences and uncheck the auto-center parts checkbox? If so, then it is most likely a duplicate of issue #11 and it has already been fixed.

Eriobis, 2016-11-04:
Nop ... the problem is the same.
I tried with other file ( the treefrog by example ), it has the exact same behavior. At least the bug is easy to reproduce :P  I only have to load a file  hehe

bubnikv, 2016-11-04:
> The software hang for a while, until I see the perl interpreter crash
window.

What does the crash window say?

We had a customer with a laptop with a dual graphics card. Slic3r refused
to run, if the 2nd graphics card was enabled. Once he disabled the 2nd
graphics card, Slic3r ran.

Eriobis, 2016-11-04:
I attached the dump file on the 1st post.
[image]

bubnikv, 2016-11-04:
I just found out there is an issue with the newest Strawberry Perl and
exception handling. I have to set up the build server to use an older
version of Strawberry Perl for building the 64bit Slic3r builds. You can
try our 32bit Slic3r build, whether it behaves better. Likely it will not
slice, but it will display some explanation why it does not.

Eriobis, 2016-11-07:
Okay sorry for the delay,

The 32 bit version works fine for the moment.
I will load some heavy file tonight, for the moment the treefrog works good.

I will close the issue since there's a workaround

closed 2016-11-07 by Eriobis