prusa3d/PrusaSlicer #5: Errors Compiling 1.31.2 on LinuxMint

voodoo-bravo, 2016-11-02:
### Version
_1.31.2 from https://github.com/prusa3d/Slic3r/releases/tag/version_1.31.2_

### Operating system type + version
_LinuxMint 17.3, fresh install, all Updates installed_
_Perl 5.24 via Perlbrew_
```
markus@mint17 ~/Slic3r/Slic3r-version_1.31.2 $ perl -V
Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=linux, osvers=3.19.0-32-generic, archname=x86_64-linux
    uname='linux mint17 3.19.0-32-generic #37~14.04.1-ubuntu smp thu oct 22 09:41:40 utc 2015 x86_64 x86_64 x86_64 gnulinux '
    config_args='-de -Dprefix=/home/markus/perl5/perlbrew/perls/perl-5.24.0 -Aeval:scriptdir=/home/markus/perl5/perlbrew/perls/perl-5.24.0/bin'
    hint=recommended, useposix=true, d_sigaction=define
    useithreads=undef, usemultiplicity=undef
    use64bitint=define, use64bitall=define, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cc', ccflags ='-fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64',
    optimize='-O2',
    cppflags='-fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include'
    ccversion='', gccversion='4.8.4', gccosandvers=''
    intsize=4, longsize=8, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=define, longlongsize=8, d_longdbl=define, longdblsize=16, longdblkind=3
    ivtype='long', ivsize=8, nvtype='double', nvsize=8, Off_t='off_t', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='cc', ldflags =' -fstack-protector -L/usr/local/lib'
    libpth=/usr/local/lib /usr/lib/gcc/x86_64-linux-gnu/4.8/include-fixed /usr/include/x86_64-linux-gnu /usr/lib /lib/x86_64-linux-gnu /lib/../lib /usr/lib/x86_64-linux-gnu /usr/lib/../lib /lib
    libs=-lpthread -lnsl -ldl -lm -lcrypt -lutil -lc
    perllibs=-lpthread -lnsl -ldl -lm -lcrypt -lutil -lc
    libc=libc-2.19.so, so=so, useshrplib=false, libperl=libperl.a
    gnulibc_version='2.19'
  Dynamic Linking:
    dlsrc=dl_dlopen.xs, dlext=so, d_dlsymun=undef, ccdlflags='-Wl,-E'
    cccdlflags='-fPIC', lddlflags='-shared -O2 -L/usr/local/lib -fstack-protector'

Characteristics of this binary (from libperl): 
  Compile-time options: HAS_TIMES PERLIO_LAYERS PERL_COPY_ON_WRITE
                        PERL_DONT_CREATE_GVSV
                        PERL_HASH_FUNC_ONE_AT_A_TIME_HARD PERL_MALLOC_WRAP
                        PERL_PRESERVE_IVUV USE_64_BIT_ALL USE_64_BIT_INT
                        USE_LARGE_FILES USE_LOCALE USE_LOCALE_COLLATE
                        USE_LOCALE_CTYPE USE_LOCALE_NUMERIC USE_LOCALE_TIME
                        USE_PERLIO USE_PERL_ATOF
  Locally applied patches:
	Devel::PatchPerl 1.38
  Built under linux
  Compiled at Nov  1 2016 23:26:11
  %ENV:
    PERLBREW_BASHRC_VERSION="0.76"
    PERLBREW_HOME="/home/markus/.perlbrew"
    PERLBREW_MANPATH="/home/markus/perl5/perlbrew/perls/perl-5.24.0/man"
    PERLBREW_PATH="/home/markus/perl5/perlbrew/bin:/home/markus/perl5/perlbrew/perls/perl-5.24.0/bin"
    PERLBREW_PERL="perl-5.24.0"
    PERLBREW_ROOT="/home/markus/perl5/perlbrew"
    PERLBREW_VERSION="0.76"
  @INC:
    /home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/site_perl/5.24.0/x86_64-linux
    /home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/site_perl/5.24.0
    /home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux
    /home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0

```

_Ubuntu 14.04 prerequsites as stated here: 
 https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux#build-slic3r_

### Behavior
[... 74 lines cut ...]
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/libslic3r/Line.o src/libslic3r/Line.cpp
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/admesh/stl_io.o src/admesh/stl_io.c
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/libslic3r/PrintRegion.o src/libslic3r/PrintRegion.cpp
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/libslic3r/BoundingBox.o src/libslic3r/BoundingBox.cpp
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/polypartition.o src/polypartition.cpp
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/slic3r/GUI/GUI.o src/slic3r/GUI/GUI.cpp
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/admesh/connect.o src/admesh/connect.c
cc -I/home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/x86_64-linux/CORE -fPIC -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -o src/libslic3r/Fill/FillPlanePath.o src/libslic3r/Fill/FillPlanePath.cpp
In file included from src/libslic3r/Fill/FillPlanePath.hpp:8:0,
                 from src/libslic3r/Fill/FillPlanePath.cpp:5:
src/libslic3r/Fill/FillBase.hpp: In static member function ‘static coord_t Slic3r::Fill::_align_to_grid(coord_t, coord_t)’:
src/libslic3r/Fill/FillBase.hpp:111:32: error: ‘assert’ was not declared in this scope
         assert(aligned <= coord);
                                ^
error building src/libslic3r/Fill/FillPlanePath.o from 'src/libslic3r/Fill/FillPlanePath.cpp' at /home/markus/perl5/perlbrew/perls/perl-5.24.0/lib/5.24.0/ExtUtils/CBuilder/Base.pm line 174.
FAIL
! Installing ./xs failed. See /home/markus/.cpanm/work/1478048351.20832/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

```

Any Idea how this can be fixed?

Thanks, 
Markus

bubnikv, 2016-11-02:
Markus,

there seems to be difference in compilers, how they handle uninstantiated inline methods, so this problem did not trigger on any of our systems.

You need to add following line at the start of FillBase.hpp
#include <assert.h>

Thanks for the report, I also pushed the fix to the master.
Vojtech

closed 2016-11-02 by bubnikv

voodoo-bravo, 2016-11-02:
Hi Vojtech, 
thanks for your reply. 

I cloned the Repo and tried to compile again. 
As the same error occured again, I had to add `#include <assert.h>` to 4 more files. 

xs/src/libslic3r/GCode/Analyzer.cpp
xs/src/libslic3r/GCode/PressureEqualizer.cpp
xs/src/libslic3r/MultiPoint.cpp
xs/src/libslic3r/PolylineCollection.cpp

After that it compiled just fine.

Greetings, 
Markus