prusa3d/PrusaSlicer #6: Can't locate object method "SetAppDisplayName"

voodoo-bravo, 2016-11-02:
### Version
_1.31.2_

### Operating system type + version
_LinuxMint 17.3, fresh install, all Updates installed_
_Perl 5.24 via Perlbrew_
_Ubuntu 14.04 prerequsites as stated here: 
 https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux#build-slic3r_

### Behavior
_After fixing #5 I could build Slic3r without errors, building with `--gui` also worked flawlessly._
_Calling Slic3r from cli works as expected, except that the Gui won't start and Slic3r exits with the following Error_

```
markus@mint17 ~/Slic3r/Slic3r-3a31d37_compile_fix $ ./slic3r.pl 
Threading disabled
Can't locate object method "SetAppDisplayName" via package "Slic3r::GUI" at /home/markus/Slic3r/Slic3r-3a31d37_compile_fix/lib/Slic3r/GUI.pm line 95.
```

Greetings, 
Markus

bubnikv, 2016-11-02:
Markus, just comment this method call out.
It is probably not being implemented by your wxWidgets.

bubnikv, 2016-11-02:
What wxWdigets you are running on? SetAppDisplayName shall be in 3.0

voodoo-bravo, 2016-11-02:
The Ubuntu 14.04 prerequisites suggested to use libwx in version 2.8 so thats what i was running. 
After uninstalling all 2.8 Packages, Installing the 3.0 ones (that are also available via the Mint-Repos) and recompiling Slic3r, the GUI starts as expected. 

Thanks for your support. :)

closed 2016-11-02 by voodoo-bravo

bubnikv, 2016-11-02:
Thanks Markus for your effort. I shall test for the wxwidgets version and
not to call this particular function.
Vojtech