prusa3d/PrusaSlicer #7: AMF issues

jindrichbenes, 2016-11-02:
- AMF suffix has to be written manualy
- AMF does not remember scale
- AMF does not remember settings (filament, printer, quality)
- AMF renames all STL to same name (ie "spoolholder.stl" and "PSU_cover.stl" will be renamed to "spoolholder.amf" and "spoolholder.amf" after loading)
- large AMF (80 files, 200mb) does not load - error: 
Unbalanced string table refcount: (1) for "AUTOLOAD" during global destruction.
Scalars leaked: -2
Can't use an undefined value as an ARRAY reference at C:/Slic3r/lib/Slic3r/GUI/Plater.pm line 280.

CReimer, 2016-11-05:
I think we need some more infos here.
What exactly are you trying to do?
Are you talking about the stl-to-amf tool?

bubnikv, 2016-11-05:
Jindrich (Czech form of the name Henry) is a Prusa Research employee. I
know what he means. He noted the requirements to make the AMF format useful
as a general project file for the Slic3r.

bubnikv, 2016-11-05:
It is not very well documented and it is not quite clear, but Slic3r could
export the platter as an AMF file and then reimport the same file. When
exporting a platter as an AMF file, the modifier meshes and their
properties are stored into the AMF file. The idea is great, only it has not
been fully implemented yet, so Jindrich just shortly documented what needs
to be done.

lordofhyphens, 2016-11-08:
Changes for this would be useful upstream as well (IIRC there's a issue open on it there as well).

bubnikv, 2017-02-09:
> AMF renames all STL to same name (ie "spoolholder.stl" and "PSU_cover.stl" will be renamed to "spoolholder.amf" and "spoolholder.amf" after loading)

Fixed.

bubnikv, 2017-02-28:
@lordofhyphens I have implemented a C++ AMF parser, it is blazing fast.

jindrichbenes, 2017-03-07:
Everything seems to work, except for visualisation of variable layer height after AMF import. The variable layers will still generate, they are just not visualised in the Preview screen.

bubnikv, 2018-03-14:
The current master implements AMF / 3MF import / export, and both the AMF and 3MF formats now include the print/filament/printer profiles, so the AMF and 3MF formats now serve the purpose of a project file.

closed 2018-03-14 by bubnikv