prusa3d/PrusaSlicer #8: -lgl and -lglu should be upper case

flannelhead, 2016-11-02:
In `xs/Build.PL`, there are linker flags `-lgl` and `-lglu` set if built with `SLIC3R_GUI`. On Arch Linux the linker fails to find the libraries. These libraries should be capitalized: `-lGL -lGLU`. I don't know if this is different on e.g. OS X.

CReimer, 2016-11-02:
Compiling on Arch works fine. Could you test this PKGBUILD from AUR, please
https://aur.archlinux.org/packages/slic3r-prusa3d/

flannelhead, 2016-11-02:
Yes it does, but that `PKGBUILD` is not enabling the new GLEW GUI (see https://github.com/prusa3d/Slic3r/commit/2a5095a1ea6053270e6e382d52d53c07f87f5fcc and https://github.com/prusa3d/Slic3r/blob/master/xs/Build.PL#L175).

CReimer, 2016-11-02:
Ohh wow. I missed that. I'll push an updated PKGBUILD in a moment.
BTW: I can confirm this issue.

CReimer, 2016-11-03:
AFAIK MacOS is unlike Linux case-insensitive. -lGL or -lgl doesn't make any difference there.

bubnikv, 2016-11-04:
I will fix the linking, but please note that the SLIC3R_GUI compilation flag and the GLEW integration is experimental. Some day in the future I plan to perform rendering inside the C++ code instead of Perl, but not now. The SLIC3R_GUI switch currently does not enable anything useful.

bubnikv, 2016-11-04:
fixed
https://github.com/prusa3d/Slic3r/commit/3fc57ba8d85bef0ea6419cffe884e3ee151531ba

closed 2016-11-04 by bubnikv