prusa3d/PrusaSlicer #11: Slic3r crashes when adding files that don't fit the bed

uclaros, 2016-11-03:
### Version
1.31.2-prusa3d-win64

### Operating system type + version
Windows7 64 sp1

### Behavior
If 'Auto-center parts' is enabled, adding stl files that don't fit the build table cause a crash.
This happens both by adding one huge object or by adding many smaller ones.

#### STL/Config (.ZIP) where problem occurs
Any bunch of stl files that don't fit the table with 'auto-center parts' enabled.

bubnikv, 2016-11-03:
Does slic3r crash, or hang? I have seen hang ups, but not crashes.

uclaros, 2016-11-03:
Call it what you will...
[image]

chriszero, 2016-11-03:
This also happens with the latest slic3r prusa build. This happens really often to me.
Win10 64bit

uclaros, 2016-11-04:
Disabling "Auto-center parts" from preferences is a temporary solution.

bubnikv, 2016-11-04:
I am aware of the issue, spent about half a day with it two weeks before,
but I was not able to find the cause. Probably it has something to do with
the UI/computing core synchronization.

uclaros, 2016-11-04:
The bug did not exist in the stable version of upstream slic3r. I haven't tested the upstream/master branch yet.

bubnikv, 2016-11-04:
what do you mean with stable upstream Slic3r? 1.2.9?

uclaros, 2016-11-04:
Both 1.2.9  release and the upstream/stable branch, 1.2.10-dev

bubnikv, 2016-11-04:
How about the builds by @lordofhyphens?
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches

bubnikv, 2016-11-04:
All right, seems that following check-in fixed the hang-up.
https://github.com/prusa3d/Slic3r/commit/6217622865eac20a00b0a1a55dfddd2dffd51ceb

I replaced a Perl style exception from the auto arrangement function with a boolean return value. I don't understand how the old code supposed to work, maybe it was somehow dependent on versions Perl or Wx or gcc. Whatever, I replaced the code with something straightforward and easy to follow.

Closing this issue, please feel free to reopen if it reappears.

closed 2016-11-04 by bubnikv

bubnikv, 2016-11-04:
Actually the problem is Windows Strawberry Perl / MINGW specific:
If a croak() is called from a C++ try/catch block, the process hangs.
https://www.nntp.perl.org/group/perl.perl5.porters/2015/01/msg225230.html