prusa3d/PrusaSlicer #12: Bridging-Angle not optimal

mgroene, 2016-11-03:
### Version
1.30 release candidate vs. 1.31.2

### Operating system type + version
windows 7 up to date

### Behavior
Bridging in the latests release seems to get wrong. See pics.
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
Used the same given profiles in both versions from Prusa3D: 0,2mm NORMAL; Prusa PET; Original MK2 ColorPrint
[test2.zip](https://github.com/prusa3d/Slic3r/files/569926/test2.zip)

CReimer, 2016-11-03:
I think it's wrong in both pictures. Shouldn't Slic3r try to keep overhangs as short as possible? Something like in the lower picture but rotated by 90°?

mgroene, 2016-11-03:
Hm,... if it would be rotated by 90° it would try to connect to the perimeters just created and still hot. Believe that would not be as good as in the first picture, but I'm just guessing.

CReimer, 2016-11-03:
Ahh yes. That makes sense.

bubnikv, 2016-11-08:
Fixed in https://github.com/prusa3d/Slic3r/commit/22ca927f124572b08546ab3aca7b085561a0d9cb

The bridging has been broken, when I tried to fix the bridging for close regions. Now the bridging should work for both single and multiple touching regions.

mgroel, 2016-11-08:
thank you for your effort in this (single+multiple), will try again

mgroel, 2016-11-08:
Thank you for building a new release and adding further information on how to build, I didn't get it right.
Just tried the new version, looks better now, but still not perfect in every case.
[image]
Bridges in different isles get the same angle though not touching and separated by perimeters.
[image]

bubnikv, 2016-11-08:
Would you please attach the models? Thanks.

mgroel, 2016-11-08:
Uploading ZIPs with Firefox to github seems to be broken, otherwise I would have already.
Uploaded the models externally: https://oc.groene.de/index.php/s/c5kl7htxKlBesTE

bubnikv, 2016-11-08:
Thanks for very good test cases. For the current code the bridging regions are connected. I am on a good way to fix it.

closed 2016-11-10 by bubnikv