prusa3d/PrusaSlicer #13: Infill does not align with lower layers

CReimer, 2016-11-03:
### Version
Git rev f278fa454e628600e0e3c217477213070fe4193c

### Operating system type + version
Arch Linux

### Behavior
In the last stable version all infill patterns aligned with the layer below. This is no longer the case. See screenshot
[image]

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
https://www.thingiverse.com/download:90409
scaled to 1000%

bubnikv, 2016-11-03:
Thanks, I see it. That bug has been introduced while re-writing the rest of the infills from Perl to C++. I hope this applies to the bridges problem as well. The infill is aligned in the official 1.31.2 build correctly.

bubnikv, 2016-11-04:
Fixed, there was a bug in an initialization of a bounding box, it's "valid" flag has been left uninitialized, therefore the chaotic behavior.

closed 2016-11-04 by bubnikv