prusa3d/PrusaSlicer #14: Infill is not generated when "Ensure vertical shell thickness" is on

flannelhead, 2016-11-04:
### Version
_Version of Slic3r used goes here_
git master @ https://github.com/prusa3d/Slic3r/commit/3fc57ba8d85bef0ea6419cffe884e3ee151531ba

### Operating system type + version
_What OS are you using, and state any version #s_
Arch Linux

### Behavior
* _Describe the problem_
Infill is not generated when the "Ensure vertical shell thickness" option is on.
* _Steps needed to reproduce the problem_
1. Load STL
2. Select "Ensure vertical shell thickness"
3. Select infill <100 %
* _Expected Results_
Infill is expected to be seen in preview.
* _Actual Results_
Infill is not seen in preview.

The infill appears again after unchecking "Ensure vertical shell thickness" and changing infill density to trigger regeneration of infill.

Reverting https://github.com/prusa3d/Slic3r/commit/61d82b02b527066803584bc3bcfd3da852caebab fixes the issue.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[config.zip](https://github.com/prusa3d/Slic3r/files/570928/config.zip)
STL: http://www.thingiverse.com/download:648838

bubnikv, 2016-11-04:
Sorry, I cannot reproduce the problem with g3fc57ba, built from scratch on a build server for win64.
Maybe you need to do a fresh rebuild? Or is the problem specific to some sequence of steps? Would you please provide screenshots?

bubnikv, 2016-11-04:
By the way, ensure vertical wall thickness is a competing feature to the extra perimeters when needed. I think of the two as exclusive or.

flannelhead, 2016-11-04:
@bubnikv Good to know that, thanks.

I also built the mentioned revision on Windows 64 from scratch. There I couldn't reproduce the issue no matter how hard I tried..

On Linux I also built from scratch and even created an entirely new profile. Still I could reliably reproduce the issue.

Here's one screenshot that might be of interest. It seems the solid infill between the perimeter and sparse infill is generated.
[image]

bubnikv, 2016-11-04:
Thanks, I will try with Linux then.

flannelhead, 2016-11-04:
Still testing further. It appears that compiling with C++98 standard enabled (`env CFLAGS='-std=c++98' perl Build.PL`) the issue doesn't exist. So it might be something related to the `CPPVER >= 11` code.

flannelhead, 2016-11-04:
Ok, the traces seem to lead to [https://github.com/prusa3d/Slic3r/blob/61d82b02b527066803584bc3bcfd3da852caebab/xs/src/libslic3r/Surface.hpp#L100](this method). Disabling it seems to resolve the issue.

bubnikv, 2016-11-04:
Thanks a lot! Thinking about it, this construct does not make sense and the compiler has no way to know about the ownership of the pointers stored in the array. The correct solution is simply to delete this function. Thanks again.

bubnikv, 2016-11-04:
Fixed, thanks to @flannelhead 
https://github.com/prusa3d/Slic3r/commit/89cf29080b87c9d42bdeba060df1d85d5f79276b

closed 2016-11-04 by bubnikv