prusa3d/PrusaSlicer #15: Gaps between infill and perimeter leads to errors in laydown on following layer

Sebastianv650, 2016-11-04:
### Version
89cf29080b87c9d42bdeba060df1d85d5f79276b

### Operating system type + version
Win 10 64bit

Thanks for patching #3 , it's now working!
I'm afraid I'm having a new one: While the gap I got with original Slic3r is now removed thanks to enabled "Ensure vertical shell thickness", I got a new one further up in the model where the solid top layer is starting. I think the spherical, small body of the cat model is a good stress test for this feature.

What we see in the picture below is a hole between the perimeter lines, but there should be none accoring to the new feature to avoid this. So I had a look how the infill is generated in this layer and the layer before.

[image]

The following picture shows the layer before the gap. There are gaps between the infill and the perimeter, where an extra line should nearly fit. Maybe it has something todo with

> Disabled the automatic extrusion width adjustment for infills as it sometimes picked excessive values. The automatic extrusion width adjustment has a questionable value, as it really works only for rectangles parallel to the infill direction.

During the next layer, the perimeters are set inward due to the smaller diameter so that the inner of the two perimeter gets printed above this void. As this happens, it gets pulled inward because there is nothing where it can bond to while the nozzle is moving along the circle. -> Gaps is visible.

[image]

This is how such a layer is looking with original Slic3r, there are small lines that fills the gap as good as possible:

[image]

If it's the removed feature, could you add id again or make it available with a checkbox at least?

Please note that the print was done with a code base from yesterday. Now, on the screenshots it doesn't look that bad any more, might be related to the fix for #13. Neverteless, there are still more voids in the infill than with alexry's code. I will print another one and see how it behaves now.

Sebastianv650, 2016-11-04:
Confirmed also with the latest code, gaps are now smaller but also visible were there are voids in the previous infill.

bubnikv, 2016-11-04:
Sebastian,

I indeed changed something on the way the infill work. And indeed there is no simple solution to the path planning problem.

I re-wrote the rectilinear, grid, triangle, cubic infills from scratch. The infills are now very fast and they follow the perimeter line wherever possible. In the current master branch I disabled the connecting line following the perimeter for solid infills, if the line is longer than 3x the line separation distance. This leads to a better top solid fill than the build you tried before and it is close to the behavior of the old Slic3r. Actually I believe the way the solid infills worked in Slic3r at the time you printed the cat is very much the same way as it works in Simplify3D and indeed the old Slic3r achieved nicer top infill than Simplify3d. And indeed we noted that the "new" Slic3r behaves differently in regard to the top surface quality.

The original code from Alessandro contains some fudge factors, which are difficult to get hang of and the understanding comes slowly, mainly through a trial and error. 

The way the infill lines are calculated, is following: First the infill region is shrank by half a line separation. Then the shrank region is intersected with set of parallel lines spaced by the extrusion line spacing. The trimmed lines may or may not get connected. Alessandro connected the lines by straight segments, me and Simplify3D and kisslicer connect them with a piece of an offset perimeter. Now there are some slight differences. For example, Alessandro extends the lines of the infill 30% of the line separation distance than the theoretic optimum, which is similar as if you extended the perimeter-infill overlap by 30%. With this disabled in the new Slic3r, you have to set a higher perimeter-infill overlap to achieve a similar result. The Prusa3D profiles are set to 25 or 30% perimeter-infill overlap for this Slic3r Prusa Edition build.

Now why there are more gaps along the circles? First the bottom image seems to use thinner lines for the infill. I think you are mixing apples with oranges a bit. The top image shows a solid infill, which uses thicker extrusions, the bottom image shows the top infill, which uses thinner extrusions, leading to a higher chance of fitting the lines in.

The other reason why the old slic3r seems to be better, is an adaptive extrusion line width algorithm for solid infills. True, I have disabled this algorithm, as it has a questionable value in most cases. Namely, it works very well for rectangles aligned with the infill direction (which is very improbable) and now I see it works very well for circles. But that's about it. For general shapes, you will get the gaps anyway, if you don't fill them by some general over-extrusion fudge factor as the perimeter-infill overlap percentage. There is no line width adjustment factor, which would lead to a nice fill of complex shapes.

I have a version of the adaptive extrusion line width, which limits the extrusion width change to a reasonable amout, for example to +20%. I may enable it, as it should be relatively harmless and it would cover this special case of a round extrusion.

It may be reasonable to infill the "ensure vertical shell thickness" walls with a top infill to achieve finer control, but if there was a solid infill touching this "ensure verticall..." infill region, they would be filled in separately. There is no "one fits all" solution.

There are ways to improve these gaps somehow, for example by introducing a thin gap fill. It is certainly on my TODO list, but these algorithms are not simple.

I am open to a discussion on this topic. Please try to increase the perimeter-infill overlap. You may try to enable the "extra perimeters" feature for the cat model as the extra perimeters would be appropriate for circular tops.

bubnikv, 2016-11-04:
You may try to set
params.dont_adjust = true;
to false
at Fill.cpp:231

and then there is a factor_max set to 120% of the line spacing
in function 
coord_t Fill::_adjust_solid_spacing(const coord_t width, const coord_t distance)
which you may try to increase or decrease, once you set dont_adjust to false.

good luck, I am looking forward to your results, as I don't currently have time for that myself. I shall fix the bridging issues now.

Sebastianv650, 2016-11-04:
Thanks for the detailed description!

> First the bottom image seems to use thinner lines for the infill. I think you are mixing apples with oranges a bit. The top image shows a solid infill, which uses thicker extrusions, the bottom [...]

Up to now I'm using the exact same profile between the two Slic3r versions. What makes comparison a little bit time consuming is that for whatever reason only the Slic3r I built last will start, the other one throws an error message.
So all lines should be the same, I fixed everything possible to 0.48mm. Both images are showing the same (maybe +-1) layer. Alessandros Slic3r gives the solid layer, I guess because it's trying to keep the wall thickness as you try it with Ensure wall thickness.
Enabling or disabling of Extra perimeters as needed in or without a combination of Ensure vertical wall thickness isn't changing anything to the infill. In all cases, I get the zig-zag along the perimeter (yes, it is recalculating the infill when I'm checking or unchecking the boxes).

I will play with params.dont_adjust next.

Sebastianv650, 2016-11-04:
> You may try to set
> params.dont_adjust = true;
> to false
> at Fill.cpp:231

Now it looks perfect :smiley:
This is the only change I made, not a single visible fault left. From the preview, I was a little bit scared if the outermost lines are not a little bit too long, leading to over extrusion. But at least it's not visible, so I'm happy now.

[image]

Sebastianv650, 2016-11-06:
Just an idea: What's about trying to use "extra perimeters" (*) when the vertical wall thickness isn't suficient instead of using tiny zig-zag infill. If this isn't working due to a complex shape for example, use the current behaviour.
This way, especialy in the spherical example of the cat, there could be used aditional perimeters so no voids to the perimeter can occur as with the zig-zag infill and therefore the vertical wall thickness is guarantied to 100%.

*: Same name wording as the existing Slic3r option, but extra perimeters as defined at the moment isn't doing that..

bubnikv, 2016-11-08:
I have re-enabled the adaptive solid infill line width. The line width could be increased by 20% only.
https://github.com/prusa3d/Slic3r/commit/22ca927f124572b08546ab3aca7b085561a0d9cb

> Just an idea: What's about trying to use "extra perimeters" (*) when the vertical wall thickness isn't suficient instead of using tiny zig-zag infill. 

Please create a new Issue entry for this proposal as it is separate from this issue and it is really more a feature request then a bug report. Thanks.

bubnikv, 2016-11-08:
> Up to now I'm using the exact same profile between the two Slic3r versions. What makes comparison a little bit time consuming is that for whatever reason only the Slic3r I built last will start, the other one throws an error message.

You can install multiple Strawberry Perl installations on Windows. Under OSX or linux, you can use perlbrew for the same purpose.

closed 2016-11-08 by bubnikv