prusa3d/PrusaSlicer #16: Crashes while switching in preview through layers with arrow keys

CReimer, 2016-11-04:
### Version
89cf29080b87c9d42bdeba060df1d85d5f79276b

### Operating system type + version
Arch Linux 64bit

### Behavior
Switch through layers in preview screen with arrow keys. Do not touch the scrollbar on the right or it won't crash.

```Can't locate object method "set_z_high" via package "Slic3r::GUI::Plater::3DPreview" at /usr/share/perl5/vendor_perl/Slic3r/GUI/Plater/3DPreview.pm line 99.```

bubnikv, 2016-11-04:
thanks, I see I did not refactor this part of the code correctly, when I added the 2nd scroll bar.

bubnikv, 2016-11-08:
Fixed with 
https://github.com/prusa3d/Slic3r/commit/e15b5f4587c4e24d37f515d4b41bb55f958e9e52

Also added a check box to visualize a single layer only
in the 3D path view. The check box may be toggled with a 'S' key.

Added shift+U / shift+D buttons in the 3D path view to show a single
layer only.

closed 2016-11-08 by bubnikv