prusa3d/PrusaSlicer #19: Infill with transparent materials

KoKuToru, 2016-11-06:
### Version
_89cf29080b87c9d42bdeba060df1d85d5f79276b_

### Operating system type + version
_Linux (ArchLinux)_

### Behavior
* _Printing transparent material with infill is ugly_
* _Steps needed to reproduce the problem_
  * _Print with transparent material_
  * _Check visual_
* _Expected Results_
  * [image]
* _Actual Results_
  * [image]

_Is this a new feature request?_
Probably yes

----

Why is the **expected result** better for transparent materials ? 
_Because it doesn't change the width of the perimeter._

How was the **expected result** screenshot generated ?
_I quickly hacked some values..  
`FillRectilinear2.cpp@1324: bool skip = true` and   
`FillRectilinear2.cpp@1386: bool skip = true`   
That said, it looked like I might be able to get this behavior by changing the `link_max_length` value somehow.. (but 0 is not allowed)_

I might be able to create a patch for this...
_I suppose `emit_perimeter_segment_on_vertical_line` and `emit_perimeter_prev_next_segment`  generate these perimeter lines.. just need to check how to make them conditional by a config parameter.._

---- 

If the original slic3r-repro is better for this issue.. let me know I will recreate it there..
But i think the infill code here is different then the one in the original slic3r-repro..

bubnikv, 2016-11-06:
As commented in the pull request, link_max_length  shall be set to a small but nonzero value to effectively diable the infill lines fusing with the inner perimeters.

> If the original slic3r-repro is better for this issue.. let me know I will recreate it there..
But i think the infill code here is different then the one in the original slic3r-repro..

The rectilinear / grid infills are completely rewritten in the Prusa3D fork, the triangle and cubic infillls are new in the prusa3d fork.

JohnnyDeer, 2019-10-11:
This problem has already been solved.  I will close this issue in framework of improving the clarity of the issues list.

closed 2019-10-11 by JohnnyDeer