prusa3d/PrusaSlicer #21: [Request] Add the weight of filament

Eriobis, 2016-11-07:
### Version
1.31.1
_Use `About->About Slic3r` for release versions_

### Operating system type + version

### Behavior
Could you add an input in the filament section with the density so we can see
in the info screen on the plater page the weight. I'm sure i'm not the only one to use a scale to
see if I have enough filament.

_Is this a new feature request?_
Yes XD

#### STL/Config (.ZIP) where problem occurs

josefprusa, 2016-11-15:
Hi, different colors of the same material even from the same brand can ha different densities. For example natural pla and red colored one.

bubnikv, 2017-02-09:
Implemented by @lordofhyphens

closed 2017-02-09 by bubnikv

CReimer, 2017-02-09:
Can someone explain this feature?
I only get Filament usage in mm³ all other values stay 0.00. And I only get any values after saving the G-Code. That's a major inconvenience.

CReimer, 2017-02-09:
OK. I have to set some values in filament settings before. What remains is the inconvenience that the information does only show up after the G-Code is saved.

lordofhyphens, 2017-02-10:
That's because it's only calculated during the gcode export process
(because that's when everything is finally laid out and the synthesis to
exact flows, etc, is done).

Doing it earlier would involve a large refactor or more code duplication
than I'm comfortable doing.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-02-10:
I should probably add a tooltip that indicates that it is only available
post-gcode export.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.