prusa3d/PrusaSlicer #22: [Request] 'Support editor' like Simplify3D or Craftware Slicer

CReimer, 2016-11-08:
Automatic support generators often generate supports which are difficult to print and are likely to break while printing)

Having the possibility to add more support on some places and remove on other gives additional possibilities for the print.

lordofhyphens, 2016-12-06:
Any news about this? I was going to look into a partial solution using the modifier mesh interface, either through hacking the existing Perl code or porting it to XS and then having my way with it.

bubnikv, 2016-12-08:
I would like to finish the automatic supports first, then the UI. I was thinking about calculating the support volumes and somehow let the user decimate them.

Boby71, 2017-05-31:
Well, it'd be great if Slic3r could add and remove single support pillars: remove where support doesn't seem to be necessary, add additional pillars where more support seems to be needed.
But to remove "unnecessary" supports would be really a great step forward! Thanks!

lordofhyphens, 2017-05-31:
@Boby71 https://github.com/alexrj/Slic3r/issues/3063 This has been on the mainline "please do" list for quite some time. I had some thoughts as to how to go about it but it would require significant refactoring of the supports.

bubnikv, 2018-09-27:
We have support enforcers / blockers now.

closed 2018-09-27 by bubnikv