prusa3d/PrusaSlicer #23: [Request] Visualize print speeds in Preview tab

CReimer, 2016-11-08:
No Further Text

chriszero, 2016-11-08:
Did you mean overall printing time of the sliced model?

CReimer, 2016-11-08:
No. But that would be nice, too.
I'm looking for the individual speeds for the different parts. Especially when Slic3r decides to slow down the print, because a specific layer is printed too fast.

Something like what http://gcode.ws/ does. Different line colors for their specific print speed.

bubnikv, 2016-11-08:
That's also on my list, but it requires analyssis of the generated G-code, so it does not fit into the existing code base easily. When having the G-code analysis, a natural addition is the print time and filaent consumption.

bubnikv, 2018-03-14:
Implemented by @enricoturri1966 in the current master. It will be part of the upcoming Slic3r PE 1.40.

closed 2018-03-14 by bubnikv