prusa3d/PrusaSlicer #26: Strange behavior of Extra perimeters and ensure wall thickness

Sebastianv650, 2016-11-08:
### Version
e15b5f4587c4e24d37f515d4b41bb55f958e9e52

### Operating system type + version
Win10 64bit

### Behavior
I think something might be broken with the two functions extra perimeter and ensure wall thickness. Have a look at the pictures, showing always a single line near the top of a half sphere:

This first one is with both features disabled. I would expect to see two perimeters with normal infill, no zig-zag lines. But there is something:
[image]

In this one, extra perimeters are enabled. No visible change compared to the first one. Also not what I would expect:
[image]

Finaly, I disabled extra perimeters and enabled ensure vertical wall thickness. (You mentioned once that they are supposed to be an exclusive or.)
Now the zig-zag is wider than before. OK, this result looks logical:
[image]

So my question is, is there something wrong with disabled features and with extra perimeters or is it me and I'm not understanding what is the right behavior?

#### STL/Config (.ZIP) where problem occurs
A simple sphere or half sphere.

bubnikv, 2016-11-08:
Would you please attach the STL?

bubnikv, 2016-11-08:
I am afraid this is how Slic3r currently works. The zig-zag surface is a kind of "ensure vertical wall thickness" feature, which has been in Slic3r forever, but it only works with top / bottom surfaces. It adds a projection of an inflated top/bottom surface to +(n_top-1) surfaces below the top surfaces and +(n_bottom-1) surfaces above each bottom or bottom bridging surface . It only happens, that after a couple of layers there is not much left from the projection of the top surface, which in this case probably was just a thin ring. It is difficult to explain, I spent months learning the guts of the Slic3r. This behavior is not exactly nice and there are many more tiny inconveniences.

These inconveniences are in other slicers as well, but Slic3r provides a nice visualization, which makes these problems visible.

By the way, if you go to the 2d path preview, move the mouse above the graphic window and press a numeric key '2', '3' ... '5', it will show a simulation of over / underextrusion.

bubnikv, 2016-11-08:
So I believe the behavior you described is not new in the Slic3r Prusa Edition. It always worked that way.

Sebastianv650, 2016-11-08:
I added the stl, renamed to .txt so Github will not block it. 

Alexrj Slic3r produces this infill (same profile, same stl, same layer shown), regardless of enabled or disabled extra perimeters. Except for a small hole in the middle, it's already a solid layer:
[image]

bubnikv, 2016-11-08:
Thanks, it looks bad indeed.

I have added debugging output to Slic3r, which exports SVG files per slice, showing classification of regions by colors. You may enable it by setting SLIC3R_DEBUG_SLICE_PROCESSING=1 in Object.pm and also uncommenting the same in libslicer.h, then recompiling. Then create an out directory in the Slic3r directory and you will get a bunch of files generated. Slic3r will be very slow though.

I am looking into it.

bubnikv, 2016-11-08:
I think I found the block I disabled due to a bad understanding of its function. Thanks for pointing this out.

bubnikv, 2016-11-09:
Fixed by 
https://github.com/prusa3d/Slic3r/commit/eb0ab3861872a8cf1ab08b0d02a16942acdabaac
Old behavior is re-enabled when ensure vertical wall thickness is off.

closed 2016-11-09 by bubnikv