prusa3d/PrusaSlicer #28: Possible improvement for "Ensure vertical shell thickness"

Sebastianv650, 2016-11-09:
### Version
317e9131e89782c98201849a21c0e2e91c8fc142

### Operating system type + version
Win 10 64bit

### Behavior
At the moment with enabled vertical shell thickness, Slic3r begins to start a zig-zag line around the perimeter to ensure the shell thickness. I used the half sphere again for the tests, watching at the upper half where the angle gets flat:
[half_sphere.txt](https://github.com/prusa3d/Slic3r/files/581241/half_sphere.txt)

Depeding on the shape of the perimeter, this can lead to following problems:
- Especialy when there is only the need for a small additional solid infill, there are areas where a small gap is the result of the used angles for the infill line. This means, in this areas the shell thickness isn't ensured and it may lead to gaps in the layers afterward in extreme cases:
[image]
- Short zig-zag is always a bit slow because the printer has no real distance to accelerate between junction points.

Proposal:
Try to use extra perimeters in the areas where the wall thickness would be too low. With this, the print would be much faster and there is no risk of gaps due to the effect seen above. In the case of the half sphere, the Layer would look like this:
[image]

- The amount of extra perimeters could be calculated by [extra width that has to be solid infill] / [perimeter width] rounded upward for safety.
- It shouldn't be a complete perimeter if not needed, like the current behaviour of "ensure vertical shell thickness" the perimeter sould be clipped to the areas where it is needed.
Lets assume only a section of the circle has to get additional shell thickness, the infill for ensure vertical shell thickness could look like this (shown without normal infill because I'm lazy :-p):
[image]
I hope you can imagine what I mean, in this example we would have 4 extra perimeters, connected on their ends like normal infill.

I'm not sure if there are shapes where this strategy would fail so Slic3r has to fall back to zig-zag infill. At the moment, I can't think of an example, but there might be one..?

wavexx, 2019-11-03:
Adding perimeters as opposed to some rectilinear infill would definitely be a superior approach. As @Sebastianv650 says, the main issue would be to clip perimeters to the region which needs extra support. On the other hand, *all* small areas would actually benefit from a contour filling pattern as opposed to narrow rectilinear fill (faster, less over-extrusion).

The main issue I have with "ensure vertical shell thickness" is that using 1-2 perimeters on an overhanging section will often overheat the perimeter itself because too much time is spent on the overhang. This is especially true with the V6 nozzle, which has a wide flat tip. This is compounded on materials that need to be printed hot and are prone to warping.

Currently adding extra perimeters and removing "ensure vertical shell thickness" will almost always yield better results, and generally a faster print as well.

TNTINC, 2021-09-03:
This is easily my biggest gripe with prusaslicer and I'm really disappointed that it's been ignored for nearly five years.
It pains me to see my printer pointlessly shake itself apart.

PaparazziN, 2023-10-21:
With the release of PrusaSlicer 2.6.0-alpha5 the option "Ensure vertical shell thickness" was removed and an improved algorithm was implemented. Could it be assumed that we can close this issue?

lukasmatena, 2023-10-23:
@PaparazziN I agree, closing.

closed 2023-10-23 by lukasmatena