prusa3d/PrusaSlicer #29: README points to outdated build instructions

VanessaE, 2016-11-10:
...at least for Linux.  Build instructions for it are here, https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

(see also the README for Slic3r upstream)

bubnikv, 2016-11-11:
Welcome Vanessa to the Slic3r Prusa Edition.

Would you please point me to the file I shoud modify?

By the way, you are invited to enter your pressing needs into our issue tracker.

VanessaE, 2016-11-11:
The main README in your repo, i.e. this file:  https://github.com/prusa3d/Slic3r/blob/master/README.md

lordofhyphens, 2016-11-18:
This should be taken care of the next time @bubnikv merges in from upstream, as it's correct there. 

The commit is https://github.com/alexrj/Slic3r/commit/34f1853dd0bf67e24646ae7912b94a59c52e9682

bwbeall, 2016-11-21:
Hi Vanessa,

Sorry to butt in on your discussion (this is Win64 rather than Linux, but related to build instructions) but this seems like a good place to add my two cents. I have built the Win64 release version of prusa3d/Slic3r (twice now!) with these instructions. They could use some more work, but at least it was somewhat reproducible (for my setup).

Windows 10.0.14393 Pro x64
VS 2013 Version 12.0.40629.00 Update 5 (Community Edition)
NET Framework 4.6.01586

# Instructions for building slic3r for MSVC12 64-bit
## Installing the dependencies:
### Installing MS Visual Studio 2013 (MSVC12.0) Express
https://www.microsoft.com/en-us/download/details.aspx?id=48134
You may also be able to install the Community Edition of Visual Studio 2013 (for individuals and open-source), which has a few enhancements.
 
### Installing the Boost libraries:
Slic3r is known to work with Boost 1.59.0. Boost 1.61.0 changed things and thus the build instructions do not work for it.
Download them from the Boost website. The official binaries won't work since they're compiled for MSVC. Extract the sources to C:\dev\boost_1_59_0. Run the following instructions from the Perl command prompt.
MINGW instructions:
cd C:\dev\boost_1_59_0
bootstrap.bat mingw
b2 toolset=gcc link=static runtime-link=static variant=release
 
or MSVC12 instructions:
cd C:\dev\boost_1_59_0
bootstrap.bat vc12
b2 toolset=msvc link=static runtime-link=static variant=release
 
Note: This appears to build vc14, rather than vc12, but I'm not sure. In any case, it didn't work properly for me. In the end, it was much simpler to install the pre-built 64-bit Boost libraries from here (auto-download):
https://sourceforge.net/projects/boost/files/boost-binaries/1.59.0/boost_1_59_0-msvc-12.0-64.exe/download
or browse here:
https://sourceforge.net/projects/boost/files/boost-binaries/1.59.0
 
### Installing Cygwin
You will need to have various Cygwin packages installed - start here:
https://cygwin.com/install.html
I have chosen to save the setup-x86_64.exe (rather than repeatedly download it from the Cygwin site), which you will need to run every time you download a package. I accepted all of the default choices offered. These are the packages that I installed (some of these packages and many libraries were  installed by default or as prerequisites):

Archive:
bzip2
unzip
xz

Base:
All

Devel:
cygwin-devel
make
mingw64-x86_64-binutils
mingw64-x86_64-gcc-core
mingw64-x86_64-gcc-g++
mingw64-x86_64-headers
mingw64-x86_64-runtime
mingw64-x86_64-windows-default-manifest
mingw64-x86_64-winpthreads
patch

Interpreters:
gawk

Net:
ca-certificates
curl
openssl

Shells:
bash
dash
mintty

[... 209 lines cut ...]
Download slic3r from alexrj or Prusa3D:
perl Build.PL
 
This may fail in several places. My latest build using the Prusa3D master on 2016.11.21 built and ran flawlessly (many thanks to bubnikiv and lordofhyphens)!
### Install IO-stringy
If this fails, you will need to install IO-stringy. Note the folder in which IO-stringy work was performed and failed (it will be in a folder named something like "wperl64r\cpanm\work\1234567890.12345" - it looks like a date/time stamp, so the folder names are always monotonically increasing).
cd c:\wperl64r\cpanm\work\1234567890.12345
nmake install
 
Then continue building slic3r (back in the slic3r folder):
 
perl Build.PL
perl Build.PL --gui
 
### Reinstalling OpenGL-0.6704
The Build.PL in the previous step may have clobbered the version we installed, and it may need to be reinstalled. If so, go to C:\Installs\OpenGL-0.6704, and reinstall this version of OpenGL:
 
cd C:\Installs\OpenGL-0.6704
nmake install
 
### Run slic3r
 
Again, back in the slic3r folder:
 
perl slic3r.PL --gui

bubnikv, 2017-03-23:
Updated. Thanks.

closed 2017-03-23 by bubnikv