prusa3d/PrusaSlicer #31: add missing libwx libraries to the LInux builds

bubnikv, 2016-11-11:
libwx_baseu-3.0.so.0
libwx_gtk2u_adv-3.0.so.0
libwx_gtk2u_core-3.0.so.0
libwx_gtk2u_gl-3.0.so.0
libwx_gtk2u_html-3.0.so.0

CReimer, 2016-11-11:
Just took a quick look at your Linux packages. That's not how Linux packages should be made.
You can't just randomly bundle libraries like you would do on Windows.
There are also file hierarchy standards on linux.

You provide pictures in /bin --> No go
You provide libraries in /lib --> No go

That's not how things work on Linux.

I think for you it's best to focus on Ubuntu. Everyone else doesn't really care about prebuilt packages. Or even actively avoids prebuilt packages.
You already provide source code. That's usually enough for Linux.

bubnikv, 2016-11-12:
> You can't just randomly bundle libraries like you would do on Windows.

I am aware of that. I even worked for SuSE around 1999 as a package manager for half a year. We are having a Linux buildbot running, which is useful at least to verify, whether it compiles and the automatic tests run through. Having the compilates, I am giving them out, someone may get lucky to have it compatible with his runtime libraries. I know the probability is quite low.

> I think for you it's best to focus on Ubuntu.

I am in contact with @hyperair, who is a co-maintainer of the Debian Slic3r package. I hope we can come up with something. Miro Hroncok also prepaired a Fedora package, which I am to test today. Things are moving.

> You already provide source code. That's usually enough for Linux.

Not quite with the Perl mess.

CReimer, 2016-11-12:
> I even worked for SuSE around 1999 as a package manager for half a year.

Erm, to clarify this a bit. My post wasn't intended as an insult. By just looking at the packages it seems like you're currently headed in the wrong direction.

> I am in contact with @hyperair, who is a co-maintainer of the Debian Slic3r package. I hope we can come up with something. Miro Hroncok also prepaired a Fedora package, which I am to test today. Things are moving.

For Arch Linux there's a build script available here: https://aur.archlinux.org/packages/slic3r-prusa3d/
I'm also maintainer of this build script.

> Not quite with the Perl mess.

I wasn't aware of that until I read about your problems on Windows in one of the other issues. For Arch Linux this wasn't a problem. Everything works there quite good.

But I agree with you. Perl is already a mess when using XS with C. But it gets even worse with C++.

bubnikv, 2016-11-12:
> Erm, to clarify this a bit. My post wasn't intended as an insult. By just
looking at the packages it seems like you're currently headed in the wrong
direction.

I know.

Thanks for helping with the Arch linux. The linuxes are all over my head.

bubnikv, 2016-11-20:
I wonder whether the official alexr binary builds for Linux are all right in regard to binary compatibility to various linux distributions.

bubnikv, 2016-11-20:
>I wonder whether the official alexr binary builds for Linux are all right in regard to binary compatibility to various linux distributions.

To answer myself, I tested the alexrj 1.2.9 slic3r on Fedora 24 Workstation 64bit and it worked all right. So while this path is not quite kosher, it will likely work for many and it is threfore worth IMHO.

bubnikv, 2016-11-20:
On ArchLinux, the alexrj 1.2.9 build required to install gtk2, glu & glut libraries. Other than that, the binary alexrj Slic3r runs fine.

hyperair, 2016-11-21:
On Sun, Nov 20, 2016 at 05:06:06AM -0800, bubnikv wrote:
> On ArchLinux, the alexrj 1.2.9 build required to install gtk2, glu & glut
> libraries. Other than that, the binary alexrj Slic3r runs fine.

I've done up the slic3r-prusa packaging and uploaded it to Debian. It's waiting
in the NEW queue[0] as it's a new package that has never been in Debian before.
The wait period can range between a couple of weeks and a few months.

The requisite libraries are expressed as dependencies and automatically
installed together when you apt-get install slic3r-prusa, so please avoid
bundling libraries in if possible. If you feel that it is necessary to, I'll
need to generate a stripped tarball to remove those dependencies when packaging
for Debian, which just makes the downstream delta grow.

Debian Packaging can be found at [1].

[0] https://ftp-master.debian.org/new.html
[1] https://anonscm.debian.org/git/3dprinter/packages/slic3r-prusa.git

-- 
Kind regards,
Loong Jin

bubnikv, 2016-11-22:
Thanks @hyperair for your work indeed.

> The requisite libraries are expressed as dependencies and automatically installed together when you apt-get install slic3r-prusa, so please avoid bundling libraries in if possible. If you feel that [...]

Thanks for the notice. I received a similar comment from the Fedora maintainer. I will ponder over it as bundling the libraries makes my life easier on Windows and OSX, and it makes easier for people to compile even on linuxes, as I cannot guarantee, that my build scripts will be compatible with all linux distributions. 

The libraries I bundled are tiny, so I don't really see a pressing need to remove them from the source distribution, but I understand your urge (and the urge of Linux distribution maintainers) to avoid static linking. I am considering to add a build script switch to switch between the bundled and external libraries.

Thanks again,
Vojtech

hyperair, 2016-11-22:
On Tue, Nov 22, 2016 at 12:27:39AM -0800, bubnikv wrote:
> Thanks @hyperair for your work indeed.
>
>
> Thanks for the notice. I received a similar comment from the Fedora
> maintainer. I will ponder over it as bundling the libraries makes my life
> easier on [...]

https://wiki.debian.org/UpstreamGuide might have some useful tips. I recommend
reading it.

-- 
Kind regards,
Loong Jin

bubnikv, 2016-11-28:
Closing this issue, we have a "binary" package which seems to work on modern distros, and there is a Debian package on the way.

closed 2016-11-28 by bubnikv