prusa3d/PrusaSlicer #32: Thin wall: Gapfill not working

Sebastianv650, 2016-11-12:
### Version
c1af9bda0a9fcf2703c5847fbd9d43c62c7f9c6c

### Operating system type + version
Win 10 64bit

### Behavior
Even with detect thin walls enabled, Slic3r tries to fill the thin walls of the model in the upper area with 2 perimeters, that would result in masive over-extrusion. A few layers down, where the wall thickness is a little bit lower, it creates 1 perimeter with a gapfill-infill as desired.
3D Preview of the over-filled layer, you can baraly see the 2 perimeters:
[image]
2D View of the layer:
[image]

Alexrj's Slic3r did a good job on the model, same layer, same settings:
[image]

It looks to me as in prusa3d-Slic3r the gap fill line width is limited to a max. value. If that line width is exceeded, it tries to use normal perimeters without a check that this can't work?
Excelent gapfill of thin walls is the main reason I love Slic3r, so it would be great if this version can keep this feature at the same quality :-)

Another thing I noticed during slicing the model: alexrj Slic3r says Manifold: yes. Prusa Slic3r is stating over 800 errors in the model? Might be related, but maybe not..

#### STL/Config (.ZIP) where problem occurs
http://www.thingiverse.com/thing:1817180

bubnikv, 2016-11-12:
Which version of Alexrj's Slic3r are you referring to?

Sebastianv650, 2016-11-12:
The latest code from the Master branch: 8910b4970a18127199e29e71be5278b7abcbdad3
I only added some PRs to fix autocooling and other small things that are broken in the current version.

bubnikv, 2016-11-12:
Thanks. From the top of my head I am aware of no change, which could lead
to such a behavior, but I will dig into it.

closed 2016-11-12 by bubnikv

bubnikv, 2016-11-12:
Thanks, I have reverted a recent commit, which causes this behavior. I did not quite understand the code, now I believe I do much better. Learning by a trial and error, but I am slowly getting confident with the Slic3r internals. It took quite a long time indeed. It is also an answer, why there are not many contributors to Slic3r, as it takes quite some time to work in.

We at Prusa3D have the "detect thin walls" off as it introduces unnecessary roundings of the perimeters, which then need to be filled in by the not very reliable gap fill. There is still room for improvement indeed.

Thanks for keeping watch over this fork and for doing a great testing job. I hope with this commit we can release a useful build to the world.

bubnikv, 2016-11-12:
> Another thing I noticed during slicing the model: alexrj Slic3r says
Manifold: yes. Prusa
> Slic3r is stating over 800 errors in the model? Might be related, but
maybe not..

I have no good explanation for that.

Sebastianv650, 2016-11-12:
> It took quite a long time indeed. It is also an answer, why there are not many contributors to Slic3r, as it takes quite some time to work in.

Absolutely true! I would love to contribute more actively to Slic3r, but up to now I found no starting point. I think you need real passion to dig into how slicers work..
And I'm very glad you are continuing alexrjs work in such a activ maner since it looks he retired from development at the moment.

> We at Prusa3D have the "detect thin walls" off as it introduces unnecessary roundings of the perimeters, which then need to be filled in by the not very reliable gap fill. There is still room for improvement indeed.

Might be true, but it's the best available at the marked. :simple_smile: My experience with it is quite good, the parts I'm printing are 50% impossible to slice with Cura or Simplify3D due to thin elements.

I tested your recent change, it's nearly working now. See the following screenshots at layer height 35.8mm, it's now creating the gap fill but does a zig-zag infill also. 2D view also shows this:
[image]
[image]
If this can be solved also, it would be perfect.

bubnikv, 2016-11-12:
Is the double infill bug away if you disable the "ensure verticall wall
thickness" feature? It may be, that it does not take the gapfill into
account. Thanks.

Sebastianv650, 2016-11-12:
This pictures are without ensure vertical wall thickness, i have not tried enabling it with this model. But I will test it..

bubnikv, 2016-11-12:
thanks, that enough.

Sebastianv650, 2016-11-12:
Happens with and without ensure vertical wall thickness.

bubnikv, 2016-11-12:
I fixed a bug I have introduced recently (again). Thanks for pointing this out.