prusa3d/PrusaSlicer #34: [Question] Which thickness is ensured by "ensure vertical wall thickness"

Sebastianv650, 2016-11-16:
This is only a question, nothing is working wrong :)
I was reading your last commit regarding the vertical wall thickness and noticed that it depends on the number of top/bottom solid layers count. Up to now, I thought it will use the thickness a real vertical wall has (number of perimeters * line width).

Can you explain if there is a specific wall thickness that is ensured, like "numbers of top solid layers * layer height" or is it some % calculation?
I guess overhangs are calculated with the number of bottom solid layers, inward slopes are using the top solid layers?

bubnikv, 2017-02-02:
It ensures a number of layers above / below the current layer dictated by the number of the top / bottom layers. It behaves the same way in this regard as the CuraEngine 15.04. If you think it could be improved, please propose something.

We were discussing about the possibility to define the number of top / bottom layers in millimeters instead of the number of layers. Something like entering a percent at the end of the field means percent of something, I can imagine entering a "mm" suffix to indicate a height in mm not in the number of layers.

togomedia, 2017-03-18:
Defining the total top and bottom thickness in mm (or at least mm covered from layer thickness x number of layers) is very important when using the variable layer high feature. If a model is defined with 5, .2mm total top thickness and the layer hight is then, reduced, it is possible to get extremely thin total top thicknesses that are extremely small. 

It would be much better is the total top thickness (and bottom thickness) were preserved even when varying the individual layer height at this points. This could be done by increasing the number of layers defined for the top and bottom when the variable layer height is decreased at the top or bottom of a model.

bubnikv, 2017-06-06:
We now have a request for defining the wall thickness in mm.
https://github.com/prusa3d/Slic3r/issues/81
Let's close this issue and leave the other open.

closed 2017-06-06 by bubnikv