prusa3d/PrusaSlicer #36: Slic3r crashes when the Octoprint discovery dialog should open

bubnikv, 2016-11-18:
http://shop.prusa3d.com/forum/software-f13/new-version-of-slic3r-t2324.html

bubnikv, 2016-11-19:
> I'm using macOS which is all updated to the latest version. I have a AirPort Extreme AC providing my network connection. The Octopi connected to the Printer is accessible at pursa.local and of course it's IP address too. Entering the details in to slic3r manually either IP or pursa.local works just fine, the search option causes slic3r to crash.

Thanks, I will look into it once I get a hold of some Bonjour capable IP router, which quite hard to get hold of. Normally if you issue a Bonjour discovery packet on a network, nothing happens as this packet is not understood by the router. So it is hard to debug.

Vojtech

fiveangle, 2017-01-16:
Reason above seems odd, since no IP router is required.  One can simply setup Internet sharing on their WiFi network interface and associate the RPi with it.

bubnikv, 2017-01-27:
@fiveangle 
> Reason above seems odd, since no IP router is required. One can simply setup Internet sharing on their WiFi network interface and associate the RPi with it.

It certainly does not work if I have the R-PI connected to a different segment of a network, if the network does not support Bonjour.

fiveangle, 2017-03-24:
Newly released Slicer 1.3.3-8 still does not resolve

bubnikv, 2017-03-30:
@fiveangle by the way, which system? Linux / OSX or Windows32 or Windows 64bit?

bubnikv, 2017-03-30:
Tested on Windows and Debian Linux. Works on Windows, crashes on Debian. Likely some dependencies are missing in the binary package.

It only crashes if there are some Bonjour devices on the network. It works, if there are none.

bubnikv, 2017-03-30:
Added Net/DNS/RR/A.pm and AAAA.pm to the Linux binary distribution.
Binaries will be released soon.

fiveangle, 2017-03-30:
@bubnikv - My comment was in response to this thread specifically:

> @bubnikv 
>> I'm using macOS which is all updated to the latest version.

So didn't stipulate the OS, but it is Mac OS 10.12.3 ("Sierra", which is 64-bit only).  I definitely have other Bonjour enabled and active devices on the network.  Host and RPi connected to same class C network segment running WPA2 on Apple Airport Extreme AC with direct WiFi communication allowed.  One thing of note is SSID is Unicode string with double-byte charicters and not simple ASCII, but I don't think that is relevant in this context.

-=dave

bubnikv, 2017-03-30:
Would you please try the 1.34.0? I verified the issue is fixed for Linux at
least.

fiveangle, 2017-04-08:
1.3.4.1 - bombs away...

fiveangle, 2017-04-14:
Awaiting new build

fiveangle, 2017-06-07:
@bubnikv - This appears fixed with Prusa 1.35.2

fiveangle, 2017-06-07:
Although, I will say it takes an awefully long time (approx 10s) and triggers the pinwheel of death for the last few seconds before presenting the octoprint instance selection dialog.

bubnikv, 2017-06-07:
I just verified that the binary distribution of Slic3r-1.34.1 contains the same set of  files as the Slic3r-1.35.2, and I am not aware of any pertinent change to the Octoprint handling between 1.34.1 and 1.35.2, but I will happily close this issue :-)

Anyway, there is an issue in regard to OctoPrint and HTTPS. Maybe it has something to do with your case?
https://github.com/prusa3d/Slic3r/issues/294

bubnikv, 2017-06-07:
> I just verified that the binary distribution of Slic3r-1.34.1 contains the same set of files as the Slic3r-1.35.2, 

I meant they contain the same set of Perl modules and native libraries.

closed 2017-06-07 by bubnikv

fiveangle, 2017-06-07:
If it crashes again will resubmit.