prusa3d/PrusaSlicer #37: Slic3r does not store the OctoPrint settings into the config.ini file

bubnikv, 2016-11-18:
from http://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/drivers-1-7-5-rc2-with-slic3r-prusa-edition-update-t2322-s10.html

> just one thing i use octoprint and slic3r works great with it but slic3r dos not save the ip or api when closing down
i have sort of done a work a round save a confi file with it in and load this file when you load slic3r not very good way round
as slic3r should save it automaticly when you have input the ip and api

callimero, 2016-11-19:
I am using the build from the *.zip on Win 8.1/64bit and as I am still waiting for my mk2 I am using it with my Neo and my Velleman K8200 and it perfectly remembers the octoprint adress, Port and API Key here.

bubnikv, 2016-11-19:
Thanks for heads up.

bubnikv, 2017-03-30:
I think the guy did not store into the Printer settings.

closed 2017-03-30 by bubnikv