prusa3d/PrusaSlicer #38: Zooming not possible

mgroel, 2016-11-18:
### Version
1.31.5

### Operating system type + version
Win7 64bit

### Behavior
After some time of arranging STLs on the plater, checking the preview, switching back and forth, getting through the layers it is not possible to zoom by the mousewheel any more.

Is this just on my system or can somebody reproduce? Don't know exactly yet, how to reproduce. It seems, the mouse-wheel-input is bound to the preview layer-slider once activated.

Kolous, 2016-12-11:
Confirm. It happens sometimes but couldn't find out what it could trigger.

bubnikv, 2017-02-22:
Could you guys be more specific. 

> After some time of arranging STLs on the plater, checking the preview

Which preview?

> switching back and forth

Between which views?

> getting through the layers

At which view? How do you "get trhough the layers"?

>  it is not possible to zoom by the mousewheel any more.

At which view?

Would you please provide a description of how to reproduce the issue?

Thanks

PJR-slo, 2017-02-22:
I had this also. Then I realised that in Layer View, the scroll wheel works in reverse and will only initially let me zoom in by rotating the top of the scroll wheel backwards (in the opposite direction to the other views).

After slicing and zooming in (backwards) in the layer view, the 3D and 2D views will not zoom.

Peter

bubnikv, 2017-03-23:
@PJR-slo @mgroel @Kolous 

I am still not quite sure I understand the issue. 

Does it concern the 2D print path preview only? 

Or does it concern the 3D print path preview as well?

Is it really only the question of the direction of the scroll wheel? It may be a result of the confusion between the OSX way and the rest of the world, as on OSX the "natural" scrolling means exactly the other direction to Windows scrolling.

PJR-slo, 2017-03-23:
Try this:

Open Slic3r and load a model (3D view).  Press the "Slice now" button.  While the program is slicing, move the scroll wheel; nothing will happen.  The scroll wheel will then no function ad all, even with the model deleted and re-loaded.

In "Layer" view, the scroll wheel (on Windows) works in the wrong direction.

Peter

WhiteDragonD, 2017-03-23:
Is this the same as:  https://github.com/prusa3d/Slic3r/issues/178  

?

PJR-slo, 2017-03-23:
Same problem, same solution, different cause.

It's not an issue for me as I don't use Slic3r.  Just here trying to assist as requested.

Peter

bubnikv, 2017-03-30:
> Open Slic3r and load a model (3D view). Press the "Slice now" button. While the program is slicing, move the scroll wheel; nothing will happen. The scroll wheel will then no function ad all, even with the model deleted and re-loaded.

@PJR-slo I cannot reproduce it. I hope you have the mouse cursor hovering over the 3D view.

bubnikv, 2017-03-30:
Guys, I would be most happy to help, but I don't understand the issue. Would you please provide me with a detailed set of instructions?

Apart from the scroll direction being inconsistent, what is the issue?

Well, maybe it is Windows 7 specific? I shall test on my old home laptop then.

mgroel, 2017-03-30:
Still an issue for me, too and I'm glad you want to help here. Unfortunately, I still can't reproduce systematically and name the exact steps which work for sure to make this issue happen. Though it happens quiet often. Anyway, I don't have to push the Slice-Now button for make it happen (background-slicing is disabled, too).

If my zoom is frozen on the plater (3D-Tab) I work around that by switching to the Preview-Tab (which can be empty or full if sliced already). There I can zoom and switch back to the 3D-Tab, which takes over the zooming level gained in preview-tab.

mgroel, 2017-03-30:
Indeed it seems #178 and https://github.com/alexrj/Slic3r/issues/2424 are very similar.
I have auto arrange disabled but use the arrange button from time to time. Will test this evening and report whether its relating to that?

jindrichbenes, 2017-03-30:
I used to have this issue, but havent noticed it in the latest slic3r.

mgroel, 2017-03-30:
Could reproduce it now. If pushing the Arrange-Button the zoom functionality is disabled reproducably.
It remains sticked, see screenshot.  Maybe this happens for other buttons, too?
After switchting to preview-tab back to the 3d-tab it works normal.
[image]

mgroel, 2017-03-30:
Hmm, actually every button (arrange, slice now, export g-code, layer editing, ...) seems to stick and disabling zooming.

additionally in layer editing mode, the mouse wheel doesn't expand or shrink the size of influenced layers - at least in my installation. I'm currently on 1.33.8

mgroel, 2017-04-11:
@bubnikv works for me now in the latest release! thank you very much

bubnikv, 2017-05-20:
@PJR-slo Peter, is this issue gone with the Slic3r 1.34.1? The Slic3r 1.34.1 is built differently, so maybe the wound healed itself with time.

PJR-slo, 2017-05-20:
Yeah, it seems to be fixed.  The only remaining issue is the reversed scroll wheel action in the "layers" view only.

Peter

bubnikv, 2017-05-20:
Thanks, Peter. Happily closing this issue.

closed 2017-05-20 by bubnikv