prusa3d/PrusaSlicer #40: Cubic Infill at bigger layer heights wrong extrusion with

callimero, 2016-11-24:
### Version
Slic3r 1.31.6 - Prusa Edition -win64

### Operating system type + version
win 8.1 64bit

### Behavior
The very nice cubic infill seems to be wrongly generated at higher layer heigths.
[image]

left layer heigth 0.24mm right same scene, same params, only layer heigth 0.48mm

Looks for me as if the offset from layer to layer is not calculated correctly (to big, so the upper layer is not supported from the lower layer) or the extrusion width is not calculated correctly. It must ensure that the layers offnset not too much.

I think this was also in the prusa forum but I could not find it.

bubnikv, 2016-11-24:
The layer height has to be lower than the infill line width, otherwise the layers will not stick together well. Also it is very possible that the cubic infill requires a wider infill line wrt layer heigh than the non-sloping infill patterns. I personally see a possibility of an automatic check for the appropriate infill line width.

callimero, 2016-11-25:
Yes, some kind of warning for the stupid (like me ;-) would be nice. The other way could be to ensure in the cubic algorithm that the slope will not exceed the value which is given by the extrusion width. I still wonder why slic3r tells in the popup help that it bases the default extrusion witdth on the layer height, it should say it bases on the layer AND the nozle.

bubnikv, 2016-12-05:
Marked for enhancement, we want Slic3r to notify the user about feasibility of the settings.

bubnikv, 2017-10-03:
Interestingly enough, nobody noticed until now that the cubic infill does not produce rectanguar cubes. The cubes slic3r produced until now were 30% squished along the vertical diagonal.

fiveangle, 2017-10-03:
I noticed, I imagined it would make it less strong due to lower overlap on each infill layer.  But I hadn't done any emperical testing to prove my theory, so how could I critisize ? 😬

rongith, 2017-11-18:
I spent many times to try to have good results with cubic infill with Colorfabb XT. It was only due to the flatten cubes. Thank you bubnikv.

bubnikv, 2017-11-28:
It is great to hear that everybody is happy now, and I am happy too, because I will close this issue now :-)

closed 2017-11-28 by bubnikv