prusa3d/PrusaSlicer #41: Fractional rotations ignored

ctgreybeard, 2016-11-28:
### Version
1.31.6-prusa3d

### Operating system type + version

Mac OSX 10.12.1

### Behavior

When attempting to settle a part on the bed it may be required to rotate the part slightly. But rotations in  fractional degrees (i.e. -0.5) are ignored.

Other, integral, rotations may also be ignored such as -1 where 1 or -2 are
 recognized.

#### STL/Config (.ZIP) where problem occurs

[prusa-issue.zip](https://github.com/prusa3d/Slic3r/files/616589/prusa-issue.zip)

nebbian, 2017-01-23:
I've noticed this before as well.  To get rotations of 1 degree I've had to rotate 3 degrees in 1 direction, then 2 in the other direction.  Not very user friendly.

bubnikv, 2017-02-21:
Fixed by @alexrj

bubnikv, 2017-02-21:
https://github.com/prusa3d/Slic3r/commit/6649888d1ce9293f938eeb215ddad42b0b3eafb6

closed 2017-02-21 by bubnikv