prusa3d/PrusaSlicer #42: No selection for different settings available

fpessolano, 2016-11-28:
### Version
1.31.6 prusa version

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
OSX El Capitain

### Behaviour
The list menus for the various settings are not present, I only see the four tabs without the drop down list that allows to choose between various preserved configuration.
If this is meant like this (contrary to th original slic3r), it is pretty useless!

[image]

bubnikv, 2016-11-28:
Didn't you incidentally switch to the "Simple" mode?
Go to preferences, select the "Expert" mode and restart Slic3r.

bubnikv, 2016-12-05:
Probably in simple mode. No further response received.

closed 2016-12-05 by bubnikv