prusa3d/PrusaSlicer #45: Slic3r adds super slows speed by it self

kenoro, 2016-11-29:
### Version
1.31.6
### Operating system type + version
win8.1

### Behavior
The gcode generated has some speeds that are really really slow. The speeds in the slic3r are all put in manually but still the gcode has some 3 mm/s speeds in the first layer.
[Downloads.zip](https://github.com/prusa3d/Slic3r/files/620142/Downloads.zip)

Layer 3 acording to gcode.ws was speeds as low as 0.4 mm/s

Is it a bug? or I'm just unable to configure the silc3r properly.

Thank you!

Richie76, 2016-12-01:
I´m having a similar issue with the newest version.

I´m printing parts with smaller circles which are printed awfully now - it is like a stuttering of the printhead. With version 1.3.15 this was printed as fast as it should and the circles were really nice.

Currently doing one print with the new version and after that with the older one and recording both to show the difference.

cheers, richard

kenoro, 2016-12-01:
For me you can see the speed în gcode.ws, you should be able to see it too.
And except that the size of the Goode is way bigger then in the original
Slic3r.

bubnikv, 2016-12-01:
Would you please provide me with:
the STL
the exported config file
the generated g-code?

Thanks

kenoro, 2016-12-01:
You have the in the attached zip file. The file is at this link, the cable
support:

http://www.thingiverse.com/thing:193948

I don't have a good internet fur a couple of days, if needed I will come
back on Sunday with the file

Richie76, 2016-12-01:
Hi,

please look at the 2 pictures:
[http://www.loopback.at/?attachment_id=503](url)
[http://www.loopback.at/?attachment_id=504](url)

Version of slic3r is shown above the picture.

rgds, richard

kenoro, 2016-12-01:
> Hi,
>
> please look at the 2 pictures:
> http://www.loopback.at/?attachment_id=503 <http://url>
> http://www.loopback.at/?attachment_id=504 <http://url>
>
> Version of slic3r is shown above the picture.
>
> rgds, richard

I forgot to mention that the Goode and settings are in the zip attached to
the first post

Richie76, 2016-12-01:
Ok. lowering the speed quite a bit helped ;-)

Just curious: why is not a single G2 / G3 arc command in my gcode ?

Sebastianv650, 2016-12-01:
@Richie76 if your print head stutters that's most likely not due to too slow speeds from Slic3r, especialy if lowering the speed helped. Marlin (I guess you are using that) can only handle a specific amount of lines per second, so if your stl has a extremely high resolution or you are printing too fast Marlin can't keep up with calculation for new line segments. That's the stutter.

If you use the latest RCBugFix Branch, you might also have a look at `ENSURE_SMOOTH_MOVES`. I wrote that feature to prevent stuttering which harms print quality as you have noticed.

Richie76, 2016-12-02:
@Sebastianv650 I have a prusa i3 mk2 - can i use the RCBugFix Branch ?
Since this is a firmware this will be valid for every print right ? this sounds quite cool - something like don´t worry about getting the speed right I´ll fix that for you.

Sebastianv650, 2016-12-04:
Basicaly Marlin should run on your printer. But I don't know the prusa realy well, I only read a thread on the Marlin github that at least some version of the prusa has something like an axis calibration method that is not jet in Marlin.

> this sounds quite cool - something like don´t worry about getting the speed right I´ll fix that for you.

It prevents block buffer underruns by slowing down each block to take at least 6ms of time, because that's the time Marlin needs to calculate a new block. If you try to execute shorter blocks, the buffer will run dry after some blocks. Second, it's checking if the buffer has enough runtime in it so that Marlin can update the LCD (in this time, no new block is calculated) without the risk of draining the buffer. That has the drawback that the display is frozen as long as there are too much small blocks, but in my eyes that's not as bad as having a stuttering printer = bad print result.

Richie76, 2016-12-04:
The display freezing isn't a big deal I guess since I'm using octoprint (I
need the display only for filament change).

I'll take a look into the firmware for the option.
And I'll try to print something similar from the sd card... and Check if
there at is any difference.

bubnikv, 2016-12-04:
You will not be able to run Marlin on a MK2 printer. The original Marlin
has been heavily modified to support the automatic bed leveling and the XYZ
calibration. If you use the stock Marlin, you will loose many of the Prusa
MK2 features.

Sorry, I don't have time now to look into your issue, but it seems like it
should help to set your CAD software to generate a  lower mesh resolution.

kenoro, 2016-12-05:
​Comming back to the initial question!

Can anybody tell me why Slic3r prusa adds all these slowww speed by it
self?​

-- Dipl. Ing. Marius Ioana

   +40722435580

bubnikv, 2016-12-05:
@kenoro you applied the experimental settings "max volumetric slope positive" and "max volumetric slope negative" on page Print Settings -> Speed. Please reset the two values to zero.

The feature is experimental and it is known to cause issues with gap fill. The feature is not ready for prime time yet.

kenoro, 2016-12-05:
those settings were zero.

In the mean time i thing i found the culprit: the settings for the cooling
thresholds have a settings were if the print time is smaller then a set
time it will decrease the printing time to a certain amount thus decreasing
the speeds to accomodate a minimum cooling time of certain area.

-- Dipl. Ing. Marius Ioana

   +40722435580

bubnikv, 2016-12-05:
@kenoro 
> those settings were zero.
No, in the initial posting your config.ini had these values set to 1.8.

bubnikv, 2016-12-05:
@Richie76
> And I'll try to print something similar from the sd card... and Check if
there at is any difference.

Aha, that is another story. If you are printing over USB, the communication may be the culprit.

If you still have an issue with the stuttering, please provide the STL and the config file. Otherwise I would consider this issue solved and I will close it.

m0n5t3r, 2017-01-08:
try disabling "avoid crossing perimeters": it tends to add lots of useless moves, and in some cases they consist of a meelion tiny segments that result in the print head looking like it moves very slowly; check attached files - "avoid crossing perimeters" results in ~ two times more gcode and gems like this

```
G92 E0
G1 X100.045 Y113.756 F9000.000
G1 X100.045 Y113.756 F9000.000
G1 X100.758 Y113.367 F9000.000
G1 X101.163 Y113.160 F9000.000
G1 X101.181 Y113.052 F9000.000
G1 X101.185 Y113.029 F9000.000
G1 X101.209 Y112.919 F9000.000
G1 X101.215 Y112.898 F9000.000
G1 X101.242 Y112.770 F9000.000
G1 X101.307 Y112.411 F9000.000
G1 X101.334 Y112.278 F9000.000
G1 X101.363 Y112.151 F9000.000
G1 X101.406 Y111.941 F9000.000
G1 X101.472 Y111.564 F9000.000
G1 X101.515 Y111.341 F9000.000
G1 X101.601 Y110.940 F9000.000
G1 X101.654 Y110.677 F9000.000
G1 X101.660 Y110.642 F9000.000
[goes on and on...]
```

[spirographGear19V1-noacp.txt](https://github.com/prusa3d/Slic3r/files/692269/spirographGear19V1-noacp.txt)
[spirographGear19V1-acp.txt](https://github.com/prusa3d/Slic3r/files/692272/spirographGear19V1-acp.txt)

filipgoc, 2017-01-08:
Thanks for a great demonstration @m0n5t3r 
Rather hilarious.

m0n5t3r, 2017-01-08:
I just looked at the gcode in the example provided by OP and, sure enough, layer 3 contains code very similar to the one I posted:
```
G92 E0
G1 Z0.500 F7800.000
G1 X121.305 Y92.660 F7800.000
G1 X121.305 Y92.660 F7800.000
G1 X119.884 Y93.305 F7800.000
G1 X118.877 Y94.103 F7800.000
G1 X118.634 Y93.870 F7800.000
G1 X118.295 Y93.461 F7800.000
G1 X118.201 Y93.363 F7800.000
G1 X118.123 Y93.266 F7800.000
G1 X117.917 Y92.969 F7800.000
G1 X117.761 Y92.785 F7800.000
G1 X117.316 Y92.169 F7800.000
G1 X117.204 Y91.973 F7800.000
G1 X117.199 Y91.966 F7800.000
G1 X117.039 Y91.736 F7800.000
G1 X116.973 Y91.641 F7800.000
G1 X116.945 Y91.595 F7800.000
G1 X116.849 Y91.434 F7800.000
G1 X116.660 Y91.176 F7800.000
G1 X116.523 Y90.963 F7800.000
G1 X116.419 Y90.771 F7800.000
G1 X116.306 Y90.592 F7800.000
G1 X115.839 Y89.743 F7800.000
G1 X115.775 Y89.629 F7800.000
[...]
```
this should probably be an issue in mainline slic3r, but I don't really have the time to set up isolated environments for 1.2.9, the development version, etc. (at least 1.2.9 complains about unknown config options added by slic3r prusa edition)

In my case, well tuned retraction + Marlin's LIN_ADVANCE result in almost no stringing, so I can leave "avoid crossing perimeters" disabled for substantial decreases in print time, slicing time and lack of traces from the nozzle being dragged over horizontal surfaces that aren't the last layer.

he29-net, 2017-02-02:
I can confirm the originally reported behavior as well; I'm just printing a part to test enhanced bridge handling in the Prusa3D fork and in some areas the printer suddenly slows down to sub-1mm/s speeds. In the generated gcode I can see lines like G1 F51.8875 so it is unlikely to be a communication issue (and I'm printing from a SD card anyway). Tested on the current git version (Jan-31 / c2ddf537c6).

Model, generated gcode and config are here:
[test_data.zip](https://github.com/prusa3d/Slic3r/files/748337/test_data.zip)

he29-net, 2017-02-02:
More info: when I change all speeds that are 0 (automatic) to a fixed number, the problem disappears. That would suggest that something is wrong with the automatic speed computation. Cooling settings (as mentioned in one of the posts above) have no effect.

bubnikv, 2017-02-02:
The automatic speed calculation aka constant volumetric speed is broken.
Please don't use it.

Please use our settings as a starting point.
https://github.com/prusa3d/Slic3r-settings

bubnikv, 2017-03-23:
Also there was a bug in the path generation for the wipe movements, where the wipe movements were discretised too densely. This is fixed in the current master.

So, one issue was fixed and the automatic speed & volumetric extrusion slopes are experimental and broken, please don't use them.

bubnikv, 2020-01-21:
automatic speed & volumetric extrusion slopes was disabled long time ago. Closing.

closed 2020-01-21 by bubnikv