prusa3d/PrusaSlicer #46: Arrange button doesn't do anything

mrjohnc, 2016-11-30:
### Version
1.31.6-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
When importing multiple objects at once into Prusa Slic3r the objects clip into each other, pressing the 'Arrange' button doesn't do anything.

bubnikv, 2016-11-30:
The button does not do anything if the objects do not fit on the platter.

mrjohnc, 2016-11-30:
Ah, ok, sorry, it would be very helpful if it would say 'these objects cannot be arranged as they do not fit on the platter' or something similar.

Feel free to close this issue (and pretty please make a message pop up for when objects don't fit).

Thanks

dvoelkel, 2016-11-30:
I have the same problem. 
I imported both arms of this: http://www.thingiverse.com/thing:1778856/#files
I flattened and rotated them, so that they can fit.
But slic3r did simply nothing when clicking "arrange".

bubnikv, 2016-11-30:
> thing:1778856/#files

The current auto placement code is pretty dumb and it is non-trivial to do
an optimal packing of the objects on the print bed.

You are guys right that Slic3r should display some error message, if the
auto placement fails.

dvoelkel, 2016-11-30:
packing of the objects is very tricky to code. :)
a current workaround for me is, to arrange all pieces in a 3d-designer of choice and to export a single .stl.

MartinMajewski, 2016-11-30:
Yep, this is how I do it too. It is a tedious process, though.

1. Arranging every object by hand without a quantitative helper, like a translation menu for each axis.
2. Exporting the whole assembly to .stl and giving it a meaningful description
3. Removing each part one by one. No multi-object selection for removal. No feasible shortcut on the mac (Backspace alone would be nice - CMD + Delete is a bad choice, because MacBooks don't have a dedicated delete button, so finger breaking key pressing combos have to be performed). Loooooooong distances from the object list / the object to the remove button on the top left side of the menu bar.
4. Adding the newly created .stl file
5. Reslicing everything...

Finally there...

It would help a lot if you could at least group the files inside the list view and were able to give the group a name, that is used for the gcode that is to the printer. Export as stl could also use this group name to name the resulting file. A list grouping should not be too complicated to program and I heard that Slic3r's command line already has a renaming feature.

Before adding any new features to the slicing process, Slic3r should be updated to provide a more convenient user experience in the first place.

bubnikv, 2017-03-14:
The auto-placement should be improved, as asked for by
https://github.com/prusa3d/Slic3r/issues/185

Also @alexrj proposed a task for the google summer of code
https://github.com/alexrj/Slic3r/issues/3730

Port the SVGNest library to C++: The SVGNest library provides an algorithm for polygon nesting. It's written in JavaScript and a C++ port of it would improve Slic3r's auto-arrange feature. #3237

bubnikv, 2017-06-14:
I have ported the change by @alexrj to run the automatic placement even if the components do not fit the bed. This is better than nothing.
We already have another issue https://github.com/prusa3d/Slic3r/issues/336 regarding the improvement of the automatic placement. Let's close this one.

closed 2017-06-14 by bubnikv