prusa3d/PrusaSlicer #47: Potential for filament to get stuck after cancelling print and then telling the printer to load the filament

mrjohnc, 2016-11-30:
After cancelling a print and immediately asking the printer to load filament the printer will attempt to load filament whilst the temperature is dropping. This can mean that the hot end temperature can fall below the temperature needed to load filament whilst the printer is loading filament meaning it will cause a jam. 

I found this out when I started a print and the realising I hadn't loaded the filament so cancelled the print, this caused a jam that I have now corrected.

bubnikv, 2016-11-30:
I think this belongs to the firmware github, not to Slic3r, but thanks for
heads up anyway.

mrjohnc, 2016-11-30:
Ah, sorry, do you want me to redo it there or can you move it across?

bubnikv, 2016-12-15:
Create the issue in the firmware section, please. I cannot move the issue to another repository.

closed 2016-12-15 by bubnikv

mrjohnc, 2016-12-15:
Done

https://github.com/prusa3d/Prusa-Firmware/issues/28