prusa3d/PrusaSlicer #48: Slic3r bug in vase exterior

playlet, 2016-12-01:
### Version
1.31.2-prusa3d-win64
(https://github.com/prusa3d/Slic3r/releases/download/version_1.31.2/Slic3r-1.31.2-prusa3d-win64-full-201611011502.zip)

### Operating system type + version
Windows 10 x64 10.0.14393

### Behavior
Printed vase contains artefacts on the exterior surface in the narrowest part of the vase. Take a look at the sliced preview screenshot. STL part is smooth so this is clearly a slic3r problem.

[image]

My Prusa i3 (DIY v1, Einstein variant) starts printing this excess material mid air and then continues with the perimeters leaving the printed surface very jagged in this part of the vase. I print external perimeters first, by the way.

Vase: http://www.thingiverse.com/thing:550604
I printed the HD version STL.

My config bundle:
https://drive.google.com/file/d/0B8SUarsVNcY3cmhycDNtWVBEZEE/view?usp=sharing

I realy like this Slic3r and would like to continue using it. For the first time since I started printing, my layers are not overfilled. General Slic3r program probably has some sort of bug...

I hope you guys fix this problem. Thank you!

Best regards,

Ivan

bubnikv, 2017-01-02:
Could you please provide a config.ini, not the bundle? Also the default (the first) config in your bundle has the vase mode disabled.

I was not able to reproduce the bug with the latest source code base and with your config bundle, so I would like to test with the config, with which your Slic3r fails.

Thanks

playlet, 2017-01-02:
It was not printed in vase mode. I used 2 perimeters and no infill. You have to zoom really close to be able to see it. You can see it also in the layer view as small dots floating in the air outside the vase perimeter in about 3/4 of the vase height (narrowest part).

Please rename config.txt to config.ini (I had to rename it for the upload here).
[config.txt](https://github.com/prusa3d/Slic3r/files/681380/config.txt)

playlet, 2017-01-02:
One more thing, I can't compile latest code here on github, what perl, boost and git version do I need to install for it to work on Win10x64? Thanks!

bubnikv, 2017-01-20:
Let me turn the question around. What error message did you get?
I am setting 
set SLIC3R_STATIC=1
on the command line before running
perl Build.PL
to link boost statically.

bubnikv, 2017-01-20:
I believe this is the gap fill. The FFF printer cannot print sharp corners with a round filament. Slic3r tries to produce the sharp corners by adding thin extrusions. This only happens if you have the "Detect thin walls" feature enabled. If you disable it, the artifacts disappear.

At Prusa Research, we have this feature disabled by default as it produces the artifacts you mentioned. The feature may be useful, but not in the current implementation.

playlet, 2017-01-20:
Yes, with _detect thin walls_ disabled, everything is fine. Also, if you change nozzle diameter from 0.3 to 0.4, problem is also gone.

Thank you sir for taking the time to track down this problem. :)

closed 2017-01-20 by playlet