prusa3d/PrusaSlicer #51: [Request] VariSlice

CReimer, 2016-12-01:
I just stumbled across this video on Youtube
https://www.youtube.com/watch?v=HAmneiL5-jQ

This sound like a really nice feature and part of it seems to be already implemented. But all the nice calculation magic is missing.

Existing feature: http://manual.slic3r.org/expert-mode/variable-layer-height

CReimer, 2016-12-01:
BTW: There's already a pull request for this here: https://github.com/alexrj/Slic3r/pull/1386

CReimer, 2016-12-01:
Ahh. And now I also notice that you're already aware of this feature. There are multiple comments from you  over there.

bubnikv, 2016-12-01:
We are actively looking into this.

lonelymyp, 2016-12-03:
+1 
a very useful function!

landodragon141, 2016-12-15:
I'm really excited about this feature. This could be a potential game changer. I have no idea how to code this but I have been tossing around a theory on how you could achieve "sub-layers" in other words selectively change the layer height within a standard layer.

bubnikv, 2017-02-09:
Implemented interactive variable layer height editing.

It is quite difficult to make an automatic layer height algorithm useful for real world objects as such algorithm needs to take into account visibility and size of features, smoothness of the transitions etc. An interactive editor allows an immediate application and evaluation of the feasibility of such prints.

closed 2017-02-09 by bubnikv

bubnikv, 2017-02-10:
a teaser by Josef https://twitter.com/josefprusa/status/826752013170044933
[image]