prusa3d/PrusaSlicer #52: Boost compile error

CReimer, 2016-12-02:
### Version
latest git master

### Operating system type + version
Arch Linux x86_64 current

### Behavior
Compile error

```
cc -I/usr/lib/perl5/core_perl/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -o src/libslic3r/utils.o src/libslic3r/utils.cpp
In file included from /usr/include/boost/log/expressions/formatters/named_scope.hpp:37:0,
                 from /usr/include/boost/log/expressions/formatters.hpp:24,
                 from /usr/include/boost/log/expressions.hpp:26,
                 from src/libslic3r/utils.cpp:72:
/usr/include/boost/log/attributes/named_scope.hpp:376:44: error: macro "push_scope" passed 1 arguments, but takes just 0
             named_scope::push_scope(m_Entry);
                                            ^
/usr/include/boost/log/attributes/named_scope.hpp:413:52: error: macro "push_scope" passed 1 arguments, but takes just 0
     static void push_scope(scope_entry const& entry) BOOST_NOEXCEPT;
                                                    ^
/usr/include/boost/log/attributes/named_scope.hpp:413:17: error: variable or field 'push_scope' declared void
     static void push_scope(scope_entry const& entry) BOOST_NOEXCEPT;
                 ^~~~~~~~~~
/usr/include/boost/log/attributes/named_scope.hpp:413:17: error: expected ';' at end of member declaration
In file included from /usr/include/boost/config.hpp:61:0,
                 from /usr/include/boost/polygon/isotropy.hpp:29,
                 from /usr/include/boost/polygon/polygon.hpp:12,
                 from src/libslic3r/Point.hpp:138,
                 from src/libslic3r/Line.hpp:5,
                 from src/libslic3r/Polygon.hpp:7,
                 from src/libslic3r/ExPolygon.hpp:5,
                 from src/libslic3r/ClipperUtils.hpp:6,
                 from src/xsinit.h:54,
                 from src/libslic3r/utils.cpp:39:
/usr/include/boost/log/attributes/named_scope.hpp:413:54: error: expected unqualified-id before 'noexcept'
     static void push_scope(scope_entry const& entry) BOOST_NOEXCEPT;
                                                      ^
In file included from src/xsinit.h:38:0,
                 from src/libslic3r/utils.cpp:39:
/usr/lib/perl5/core_perl/CORE/perl.h:3528:27: error: expected identifier before '(' token
 #    define PERL_GET_THX  ((PerlInterpreter *)PERL_GET_CONTEXT)
                           ^
In file included from /usr/lib/perl5/core_perl/CORE/perl.h:5553:0,
                 from src/xsinit.h:38,
                 from src/libslic3r/utils.cpp:39:
/usr/lib/perl5/core_perl/CORE/thread.h:320:38: error: expected ')' before 'pthread_getspecific'
 #    define PTHREAD_GETSPECIFIC(key) pthread_getspecific(key)
                                      ^
/usr/lib/perl5/core_perl/CORE/thread.h:320:38: error: expected ')' before 'pthread_getspecific'
In file included from src/xsinit.h:38:0,
                 from src/libslic3r/utils.cpp:39:
/usr/lib/perl5/core_perl/CORE/perl.h:3528:46: error: expected ';' at end of member declaration
 #    define PERL_GET_THX  ((PerlInterpreter *)PERL_GET_CONTEXT)
                                              ^
In file included from /usr/lib/perl5/core_perl/CORE/perl.h:5553:0,
                 from src/xsinit.h:38,
                 from src/libslic3r/utils.cpp:39:
/usr/lib/perl5/core_perl/CORE/thread.h:324:48: error: 'PL_thr_key' is not a type
 #  define PERL_GET_CONTEXT PTHREAD_GETSPECIFIC(PL_thr_key)
                                                ^
/usr/lib/perl5/core_perl/CORE/thread.h:320:61: error: ISO C++ forbids declaration of 'pthread_getspecific' with no type [-fpermissive]
 #    define PTHREAD_GETSPECIFIC(key) pthread_getspecific(key)
                                                             ^
/usr/lib/perl5/core_perl/CORE/thread.h:320:61: error: expected ';' at end of member declaration
In file included from src/xsinit.h:38:0,
                 from src/libslic3r/utils.cpp:39:
/usr/lib/perl5/core_perl/CORE/perl.h:3528:63: error: expected unqualified-id before ')' token
 #    define PERL_GET_THX  ((PerlInterpreter *)PERL_GET_CONTEXT)
                                                               ^
In file included from /usr/include/boost/log/expressions/formatters/named_scope.hpp:37:0,
                 from /usr/include/boost/log/expressions/formatters.hpp:24,
                 from /usr/include/boost/log/expressions.hpp:26,
                 from src/libslic3r/utils.cpp:72:
/usr/include/boost/log/attributes/named_scope.hpp: In constructor 'boost::log::v2s_mt_posix::attributes::named_scope::sentry::sentry(const string_literal&, const string_literal&, unsigned int, boost::log::v2s_mt_posix::attributes::named_scope_entry::scope_name_type)':
/usr/include/boost/log/attributes/named_scope.hpp:376:13: error: 'push_scope' is not a member of 'boost::log::v2s_mt_posix::attributes::named_scope'
             named_scope::push_scope(m_Entry);
             ^~~~~~~~~~~
In file included from /usr/lib/perl5/core_perl/CORE/perl.h:5553:0,
                 from src/xsinit.h:38,
                 from src/libslic3r/utils.cpp:39:
/usr/include/boost/log/attributes/named_scope.hpp: In destructor 'boost::log::v2s_mt_posix::attributes::named_scope::sentry::~sentry()':
/usr/lib/perl5/core_perl/CORE/thread.h:320:61: error: cannot call member function 'int boost::log::v2s_mt_posix::attributes::named_scope::pthread_getspecific(int)' without object
 #    define PTHREAD_GETSPECIFIC(key) pthread_getspecific(key)
                                                             ^
error building src/libslic3r/utils.o from 'src/libslic3r/utils.cpp' at /usr/share/perl5/core_perl/ExtUtils/CBuilder/Base.pm line 174.
```

I tried bisecting this bug and commit 0d20a81354ee8ae6a73643519d718beeaeda72f4 seems to be the first time this error occurs.

bubnikv, 2016-12-05:
The master is not ready for prime time, sorry.

CReimer, 2016-12-08:
Any idea when this will be fixed again. Tried to fix it myself, but since it seems to crash inside boost I don't really know how to.

bubnikv, 2016-12-08:
I will fix it once I will be ready to publish another release, at least
internally. Until then the source base is in raw state.

CReimer, 2016-12-14:
OK now thing get weird.

```
t/angles.t ..................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/angles.t ..................... Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/arcs.t ....................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/arcs.t ....................... Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/avoid_crossing_perimeters.t .. /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/avoid_crossing_perimeters.t .. Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 1/1 subtests 
t/bridges.t .................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/bridges.t .................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 16/16 subtests 
t/clean_polylines.t ............ /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/clean_polylines.t ............ Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/clipper.t .................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/clipper.t .................... Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/collinear.t .................. /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/collinear.t .................. Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/combineinfill.t .............. /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/combineinfill.t .............. Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/config.t ..................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/config.t ..................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 1/1 subtests 
t/cooling.t .................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/cooling.t .................... Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/custom_gcode.t ............... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/custom_gcode.t ............... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 15/15 subtests 
t/dynamic.t .................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/dynamic.t .................... Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/fill.t ....................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/fill.t ....................... Dubious, test returned 127 (wstat 32512, 0x7f00)
No subtests run 
t/flow.t ....................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/flow.t ....................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 6/6 subtests 
t/gaps.t ....................... /usr/bin/perl: symbol lookup error: /home/christopher/.cache/pacaur/slic3r-prusa3d-git/src/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost3log12v2s_mt_posix14attribute_name18get_id_from_stringEPKc
t/gaps.t ....................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 1/1 subtests 
[... cut ...]
  Non-zero exit status: 127
  Parse errors: Bad plan.  You planned 27 tests but ran 0.
t/svg.t                      (Wstat: 32512 Tests: 0 Failed: 0)
  Non-zero exit status: 127
  Parse errors: Bad plan.  You planned 2 tests but ran 0.
t/thin.t                     (Wstat: 32512 Tests: 0 Failed: 0)
  Non-zero exit status: 127
  Parse errors: Bad plan.  You planned 23 tests but ran 0.
t/threads.t                  (Wstat: 32512 Tests: 0 Failed: 0)
  Non-zero exit status: 127
  Parse errors: No plan found in TAP output```

bubnikv, 2016-12-15:
The problems were caused by newly linking the boost::log library. Dynamic linking was broken due to a missing BOOST_LOG_DYN_LINK symbol. It shall be working now.

Please understand that the current top level is neither polished nor stable, though it should compile now and the test cases should all succeed.

CReimer, 2016-12-17:
```
% prove -Ixs/blib/arch -Ixs/blib/lib/ t/   
t/angles.t ..................... ok     
t/arcs.t ....................... ok     
t/avoid_crossing_perimeters.t .. /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
t/avoid_crossing_perimeters.t .. Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 1/1 subtests 
t/bridges.t .................... 1/16 /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
t/bridges.t .................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 2/16 subtests 
t/clean_polylines.t ............ ok   
t/clipper.t .................... ok   
t/collinear.t .................. ok     
t/combineinfill.t .............. /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
t/combineinfill.t .............. Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 8/8 subtests 
t/config.t ..................... ok   
t/cooling.t .................... 1/11 /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
t/cooling.t .................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 3/11 subtests 
t/custom_gcode.t ............... /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7offset2ERKSt6vectorINS_7PolygonESaIS1_EEffdN10ClipperLib8JoinTypeEd
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7offset2ERKSt6vectorINS_7PolygonESaIS1_EEffdN10ClipperLib8JoinTypeEd
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7offset2ERKSt6vectorINS_7PolygonESaIS1_EEffdN10ClipperLib8JoinTypeEd
t/custom_gcode.t ............... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 15/15 subtests 
t/dynamic.t .................... skipped: variable-width paths are currently disabled
t/fill.t ....................... /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r6offsetERKNS_9ExPolygonEfN10ClipperLib8JoinTypeEd
t/fill.t ....................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 19/19 subtests 
t/flow.t ....................... /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
/usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7diff_exISt6vectorINS_7PolygonESaIS2_EES4_EES1_INS_9ExPolygonESaIS5_EERKT_RKT0_b
t/flow.t ....................... Dubious, test returned 127 (wstat 32512, 0x7f00)
Failed 6/6 subtests 
t/gaps.t ....................... /usr/bin/perl: symbol lookup error: /home/christopher/Projects/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZN6Slic3r7offset2ERKSt6vectorINS_7PolygonESaIS1_EEffdN10ClipperLib8JoinTypeEd
[... cut ...]
  Non-zero exit status: 127
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/support.t                  (Wstat: 32512 Tests: 12 Failed: 0)
  Non-zero exit status: 127
  Parse errors: Bad plan.  You planned 27 tests but ran 12.
t/threads.t                  (Wstat: 32512 Tests: 1 Failed: 0)
  Non-zero exit status: 127
  Parse errors: Bad plan.  You planned 2 tests but ran 1.
Files=32, Tests=247,  3 wallclock secs ( 0.07 usr  0.02 sys +  3.02 cusr  0.18 csys =  3.29 CPU)
Result: FAIL
```

CReimer, 2016-12-17:
It's still exactly the same.
Here's the test with the acutal command and verbose mode enabled:
http://pastebin.com/T8b18kqS

CReimer, 2016-12-17:
If I all tests but the tests for xs. I can build everything and it also seems to work just fine.
Just these stupid tests in the t directory have issues.

bubnikv, 2016-12-17:
you have to recompile and re-link.

CReimer, 2016-12-17:
Ohh no. That's kind of embarrassing now. It's fixed. Everything works just fine...
Thanks

closed 2016-12-17 by CReimer

bubnikv, 2016-12-17:
> Ohh no. That's kind of embarrassing now. It's fixed. Everything works
just fine...

No problem. Now imagine doing it for 3 systems, allowing for both static
and dynamic Boost linking on both.