prusa3d/PrusaSlicer #53: [Request] Setting gap between the model and support in XY

lonelymyp, 2016-12-03:
Add the possibility to set the gap between the model and support in XY
Current algorithm is worthless, the gap is chosen based on the width of the support.

lordofhyphens, 2016-12-06:
There's a pull request over at the main branch @bubnikv might be able to adapt to his purposes (increasing the gap).

bubnikv, 2016-12-22:
It will be part of the new supports I am working on.

bubnikv, 2017-01-20:
There is now a "support_material_xy_spacing" parameter in the current master. The UI part was taken over from the pull request by @lordofhyphens , thanks.

closed 2017-01-20 by bubnikv

lordofhyphens, 2017-01-20:
Which were the commits that implemented this? I'd like to prepare a PR
against upstream.

bubnikv, 2017-01-20:
> Which were the commits that implemented this? I'd like to prepare a PR
against upstream.

Sorry, it has been many commits. It is only in the new C++ supports. If interested, you may just compile the latest master and play with them, but if you want to bring it to the @alexr Slic3r, it is better to wait until the feature fully stabilizes in the Prusa3D fork and then add the new files and fix the dependencies.