prusa3d/PrusaSlicer #54: Resizable application window instead of mandatory full screen

montreuilg, 2016-12-04:
### Version
1.31.6-prusa3d

### Operating system type + version
MacOS Sierra 10.12.1

### Behavior
When launched, Slic3r opens up its application window with a fixed size (full screen) that cannot be resized to desired size and placement on screen.

_Is this a new feature request?_
Yes. Please make the application window resizable, and have it remember its size and position on screen (from last session?).

Thank you,

Gabriel Montreuil

bubnikv, 2017-03-31:
> When launched, Slic3r opens up its application window with a fixed size (full screen) that cannot be resized to desired size and placement on screen.

We tested on Sierra and Slic3r window could be resized, and once closed and reopen, Slic3r opens at the last position and size on the screen. What is exactly the issue?

stelgenhof, 2018-05-21:
@montreuilg I checked this using v1.39 and v1.40 and can perfectly resize the window on OSX. Also the window size and position are maintained once closed and reopened.

tracernz, 2018-05-21:
Also not seeing this on macOS (10.13). You tried clicking the green button in the title bar?

bubnikv, 2018-05-21:
@montreuilg Maybe you can give us more detail as of what is not working as you are expecting? None of the Slic3r developers here at Prusa Research are Mac users, so we may be blind to some of the issues.

FidelCapo, 2019-01-29:
I cannot reproduce this issue on mac. Closing

closed 2019-01-29 by vojtechkral