prusa3d/PrusaSlicer #55: add a "print through octoprint" option

bubnikv, 2016-12-06:
This would have to be an end-user setting, but on the nitpicks side, I would love if there was an option to automatically start printing in OctoPrint after clicking "Send to printer". I've clicked that button more times and forgotten to go into Octoprint to start the print than I'd care to admit. (In S3D I handle this with a curl command so it's all automatic there.)

Based on what I can see in the OctoPrint REST API documentation, it's as simple as setting a variable to true:

http://docs.octoprint.org/en/master/api ... pload-file

MartinMajewski, 2016-12-06:
I would totally up-vote that feature! A better OctoPrint interaction / integration is always appreciated.

grigorig, 2017-03-16:
I think upstream Slic3r does have this feature. Maybe it can be backported.

bubnikv, 2017-11-28:
https://github.com/prusa3d/Slic3r/issues/449 mentions:
Apparently it's been implemented in Vanilla Slic3r: alexrj/Slic3r@6f6ddf5

closed 2018-06-21 by vojtechkral

bubnikv, 2018-06-21:
It will be part of the 1.40.1 release