prusa3d/PrusaSlicer #56: Slic3r 1.31.6 (mac) freezes at Preparing infill...

paingod5150, 2016-12-06:
Slic3r 1.31.6-prusa3d
MacOS 10.12.1 (16B2555)

After I add any stl file over 6mb (specifically larger dragonlock tiles) Slic3r freezes at "Preparing infill..." and pegs my processors at 100%. The app remains responsive for a bit but if left alone eventually stops responding instead of finishing.

Smaller dragonlock files slice without issue. 

Cura works fine for slicing the same stl files but I am getting top surface issues I hoped "Ensure vertical shell thickness"
[FDG0165U_DLU_Floor_4x4.stl.zip](https://github.com/prusa3d/Slic3r/files/634747/FDG0165U_DLU_Floor_4x4.stl.zip)
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/634752/config.ini.zip)

 would resolve but the freeze happens even before I select that option. Additionally it freezes Preparing infill even when infill is set to 0%.
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/634754/config.ini.zip)
[FDG0165U_DLU_Floor_4x4.stl.zip](https://github.com/prusa3d/Slic3r/files/634756/FDG0165U_DLU_Floor_4x4.stl.zip)

paingod5150, 2016-12-07:
Additional report with the same files from another user. Its the STL + Slic3r. Cura continues to print them.

bubnikv, 2016-12-07:
Thanks, I can reproduce the issue.

DrLex0, 2016-12-16:
I have the same problem with a 33MB lithophane (generated by tweaking the customizable one in Thingiverse so it outputs a 256x256 pixel resolution), this does not occur in Slic3r 1.2.9.

rob-m-andrews, 2016-12-24:
Experiencing the same issue here on my 2016 Retina MacBook Pro.

Software: Slic3r 1.31.6-prusa3d
OS: macOS 10.12.2 (16C67)

Thanks for looking into it!

bubnikv, 2017-01-02:
This issue seems to be fixed by https://github.com/prusa3d/Slic3r/commit/17db5bff8d722369b648d41644782772f285ecd0
as well.

closed 2017-01-02 by bubnikv