prusa3d/PrusaSlicer #57: Slic3r crashes on zero size values in Bed Size dialog or Configuration Wizard

klimrmad, 2016-12-09:
### Version
1.31.6-prusa3d-win32
1.31.6-prusa3d-win64

### Operating system type + version
Windows 10
Wine

### Behavior
* Slic3r crashes on Bed Size dialog ether in Configuration Wizard and Bed Size Settings dialog.
* Steps needed to reproduce the problem
  * Open any of dialogs mentioned
  * Delete content in SizeX or SizeY fields to make one of them be equal to zero.
* Expected Results
  * None as this is incorrect values.
* Actual Results
  * Native Windows - program closes immidiately.
  * Under Wine - window freezes.

#### STL/Config (.ZIP) where problem occurs
Repeatable with default config.

bubnikv, 2016-12-22:
Thanks, I will merge that for the next official release.

bubnikv, 2017-02-09:
Merged, thanks.

closed 2017-02-09 by bubnikv