prusa3d/PrusaSlicer #58: Strange supports

Kolous, 2016-12-11:
Version: 1.31.6
OS: Windows 7 64bit

Sometimes quite useless very small supports are generated. They are unprintable because because of its size and starting from layer 2.

Sorry for poor screenshots but right now can't zoom (bug #38).

[image]

[image]

Using default 0.15 config with slight modifications (brim, infill, support).
[files.zip](https://github.com/prusa3d/Slic3r/files/644332/files.zip)

Other example (default 0,15 profile just support enabled):

[image]

Kolous, 2016-12-11:
Couldn't get rid of it but with overhang treshold 44 it prints at least brim around the support base.
Resulting into:

[image]

bubnikv, 2016-12-15:
Thanks for the report. I am working on a completely new support generator, let's hope this issue will be solved by that.

Kolous, 2016-12-17:
Copy that. Looking forward ;-)

bubnikv, 2017-02-18:
The current supports in 1.33.5 should be much more stable, namely these thin threads of supports will be replaced by at least a single 'U' shaped tower if using the recti-linear support pattern with no sheath. I think I can close this issue as it mostly addresses these thin unprintable support towers.

The current algorithm is still not perfect. Please feel free to open another issue, if you find something to be improved upon.

closed 2017-02-18 by bubnikv