prusa3d/PrusaSlicer #59: First layer settings are ignored

sdjmchattie, 2016-12-11:
### Version
1.31.6-prusa3d

### Operating system type + version
MacOS 10.12.1

### Behavior
* Setting the first layer height doesn't seem to be acknowledged in the output GCode.  When I set the layer height to 0.2mm but the first layer height to 0.3mm the output GCode still goes to 0.2mm for the first layer.

On the Prusa i3, the firmware denies lowering the print head lower than 0.23mm.  This means that if the GCode instructs the printer to go to 0.2, the printer only goes down to 0.23.  The extrusions are then made as if the print head is closer than it really is and the first layer is not well bonded as a result.  It is a necessity that we can set higher first layer heights because it's impossible to produce good GCode for the printer without it.

_Is this a new feature request?_
No

bubnikv, 2016-12-15:
Sorry, I cannot reproduce the problem in Slic3r you described.

sdjm-tessella, 2016-12-15:
I'll take some screenshots later today and write out what I did and show the output gcode

sdjmchattie, 2016-12-16:
OK, so I went back to test this again and the behaviour now seems correct.  I'm rather confused because it was definitely broken, but it's working now so I can't reproduce it either.  Thanks for looking at this anyway and sorry for wasting your time.

closed 2016-12-16 by sdjmchattie