prusa3d/PrusaSlicer #60: Ensure Vertical Shell Thickness interferes with solid top layer count

Endarer, 2016-12-11:
### Version
1.31.6-prusa3d
### Operating system type + version
OSX

### Behavior
If ensure vertical shell thickness is enabled and in use for a given feature. Infill will resume after enhanced vertical wall is complete; reducing the solid top layer count.

Problem can be reproduced using the default prusa profiles on the model listed below. 
http://www.thingiverse.com/thing:1794872
In this case the solid top layer count is reduced to 1 for the middle features

First solid top layer
[image]

Second to last layer. (First layer of infill after enhanced vertical wall completion)
[image]

First solid top Layer with ensure vertical shell thickness disabled
[image]

m0n5t3r, 2016-12-22:
just got bitten by this, it kills all solid infill in some places, not just top layer, see attachment

[saintflint_mk8.zip](https://github.com/prusa3d/Slic3r/files/668614/saintflint_mk8.zip)

bubnikv, 2016-12-22:
Thanks for feedback. I will look into this issue for the next release.

bubnikv, 2017-01-02:
Fixed by https://github.com/prusa3d/Slic3r/commit/17db5bff8d722369b648d41644782772f285ecd0

closed 2017-01-02 by bubnikv