prusa3d/PrusaSlicer #61: Brim generation problem

Kolous, 2016-12-12:
Version: 1.31.6
OS: Win7 x64
Brim isn't generated for object being surrounded by another object:

[image]

bubnikv, 2016-12-15:
thanks, I am aware of this "feature" and I believe it was intentional. Maybe you could test it with the original Slic3r by Alessandro Ranelucci. I believe it works the same way. Then we may get some feedback of his reasoning when implementing this.

bubnikv, 2016-12-19:
I believe this is the same issue https://github.com/alexrj/Slic3r/issues/2026

bubnikv, 2016-12-20:
I believe it is a feature to avoid brims filling small holes. Such brims are difficult to clean. It seems there are multiple requests to make the feature smarter.

janpernik, 2016-12-20:
Probably the "min hole size" could solve the problem of filling small holes. My model I would need a brim is attached.

[podlozka.zip](https://github.com/prusa3d/Slic3r/files/664122/podlozka.zip)

supermerill, 2017-03-15:
maybe https://github.com/alexrj/Slic3r/issues/2026 can help?

bubnikv, 2017-06-19:
We shall make sure that the brim will work for concentric objects as well https://github.com/prusa3d/Slic3r/issues/365

henri9813, 2021-12-25:
Hello,

Do you haveknews about this ?

Henri

bubnikv, 2022-01-03:
Yes, implemented in PrusaSlicer 2.4.0. Closing.

closed 2022-01-03 by bubnikv

henri9813, 2022-01-03:
Ok can you link the PR which solve this issue ? @bubnikv ?

T-Qualizer, 2024-05-17:
It's still an issue. If the parts are one object, but only connected above, then there's no way to print the middle part with a brim, which causes in my case the print to fail.