prusa3d/PrusaSlicer #64: Problem with firt and last layer

DavaD85, 2016-12-17:
Version 1.31.2
OS Windows 10 pro

problema to print first layer and last layer

In these photos you can see my first layer:

[image]

and problems with my last layer

[image]

i'm printing with 0.2 normal configuration with prusa 1.75.

Sebastianv650, 2016-12-17:
Calibrate your printer so that your first layer height (printing a single perimeter) is what you set in Slic3r, at the moment you are too high above the bed. Then calibrate e steps/mm, it's way too high now.

bubnikv, 2016-12-22:
Most likely not related to the Slic3r.

closed 2016-12-22 by bubnikv