prusa3d/PrusaSlicer #65: Implement Neosanding

mrjohnc, 2016-12-17:
Pretty please implement Neosanding as described here 

https://ultimaker.com/en/community/22649-neosanding-automatic-layer-sanding

burbilog, 2016-12-18:
Wow! Really cool idea!

bubnikv, 2016-12-19:
Thanks, we will let our print expert try the process first.

m0n5t3r, 2016-12-22:
I kinda stumbled into something similar due to misundersanding percent extrusion widths, see https://github.com/alexrj/Slic3r/issues/3162 (I gave up on autospeed and percent extrusion width after that, but I'll try again with 100% when I get home)

it would be a nice addition, that surface literally felt like paper :)

m0n5t3r, 2016-12-23:
ran a quick test, 0.2mm layer height + 100% extrusion width for the top layer, with a 0.4mm nozzle; this is ngen and is more difficult to print than PLA, so there is underextrusion in some places, but the end result is not bad at all (smooth to the touch, etc.)

Regular Chinese nozzle for scale; I'm thinking getting good results with this technique will be highly dependent on machine and slicer tuning, and nozzle geometry: any extrusion width smaller than the business end of the nozzle[1] should help, but the process needs tuning. 

[1] not the hole, the external diameter - the 0.4mm nozzles I bought have virtually no meat between the hole and the taper, less than 0.1mm for sure; the 0.2mm nozzle in the photo has quite a bit
[image]

rtyr, 2020-03-22:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/458. Closing this one.

closed 2020-03-22 by rtyr