prusa3d/PrusaSlicer #66: "Support on buildplate only" produces strange artifacts

burbilog, 2016-12-18:
### Version
1.31.6-prusa3d-linux64

### Operating system type + version
Ubuntu 16.04.1 LTS

### Behavior
While slicing my object and using "Support on buildplate only" produces strange artifacts.
These things just hang in air.  It happens with any settings, even with defaults, just open my stl file, rotate it -90 on Y axis to build it upright (it prints fine that way with Cura or S3D and their "buildplate only" options) and turn on supports and bulidplate only option:

[image]
[image]

BTW, slicing this with current git version as of today (latest commit 5614e997a4bca88b29b806293cc7e548b3348211) Slic3r just hugs CPU slicing while generating supports and never finishes (at least I did not have enough patience to wait a day or two).

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[slic3r-bug-config.zip](https://github.com/prusa3d/Slic3r/files/659365/slic3r-bug-config.zip)
[helix_fan_duct_v7.zip](https://github.com/prusa3d/Slic3r/files/659366/helix_fan_duct_v7.zip)

lolorc, 2016-12-19:
also having very weird support with the same version of this slice3r fork.
and worst of all it keeps crashing, see https://gist.github.com/lolorc/94e5625ec7c0e730c64b59d0fd34ba1c

gonna check the original slic3r, but in the mean time I'm trying cura 2.4

burbilog, 2016-12-19:
The trouble is, official slic3r from git crashes on my computer at work (ubuntu 14.04.05), while prusa's release DOES work and does not crash. Weird. Problems with official slic3r began in august-september and I stopped using it and switched to Cura 2.*, but did not like it.

alexrj, 2016-12-21:
@burbilog, if you have a file and a config which crash the original Slic3r, please share them in the GitHub issue repository so that we can investigate and fix. Thank you!

burbilog, 2016-12-21:
@alexrj No file, it regularily crashed on startup, producing core files. I'll try newest git version of original slic3r, may be something has changed.

burbilog, 2016-12-21:
@alexrj  as of 2708eb0badebdb830cdc7bf30dad7b6ebf6283b4 commit, it does not crash on startup on Ubuntu 14.04.05.

bubnikv, 2016-12-22:
Thanks for feedback. I am re-writing supports completely, so there will be a new set of bugs initially :-)

burbilog, 2016-12-23:
@bubnikv Ouch :)

burbilog, 2017-02-17:
I just tried to slice the same object with 1.33.5 and now it generates another type of glitches inside the object with "build plate only" option enabled.

[image]

bubnikv, 2017-02-18:
> @burbilog I just tried to slice the same object with 1.33.5 and now it generates another type of glitches inside the object with "build plate only" option enabled.

It is not perfect, the supports leak through the walls to the other side. This is due to the way how this algorithm inflates the support islands into a rectangular grid. It is something I have on my list to improve, but I felt to better provide an imperfect build to our customers now as IMHO it is still much better than the previous one in most cases.

To avoid the leakage, you can try to:
- get rid of that piece of support which leaks by modifying the object
- decrease the grid size by changing the support line spacing.
- change the support angle
- somehow change the placement of the object against the fixed support grid

bubnikv, 2017-04-04:
It seems I have found a cure for many of the leaking support issues. It will be in the next release.

bubnikv, 2017-04-05:
Fixed with https://github.com/prusa3d/Slic3r/commit/6f5700a3a6241a56df9c7dcca88bbd05c224de98

closed 2017-04-05 by bubnikv