prusa3d/PrusaSlicer #67: Cubic infill producing >100% effective infill

tschaboo, 2016-12-19:
### Version
1.31.6-prusa3d-win644

### Operating system type + version
Win7 64bit english

### Behavior
Attached is a sample stl-file and the produced gcode.
[bug.zip](https://github.com/prusa3d/Slic3r/files/662361/bug.zip)
Beginning with the layer at about 4.20mm and up you can see how the blue infill lines parallell to the perimeter get closer and closer to each other. At layer 5.40mm there are two 0.7mm-extrusion-lines almost on top of each other. If you print the provided gcode you can see a hill forming between the perimeters as there is more material deposited than there is space!

#### STL/Config (.ZIP) where problem occurs
[Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/662372/Slic3r_config_bundle.zip)

tschaboo, 2016-12-19:
In the photos you can see the mess produced

[image]
[image]

bubnikv, 2016-12-20:
The 100% infill would rely on the material to spread from the crossings of
the lines to the holes of the pattern, which is not physically possible.
This pattern is not meant to be used for a 100% infil. The same applies for
the triangles, grid or a stars pattern.

What do you advise?

ceryen, 2017-01-31:
Could it make sense to determine the maximum sensible infill percentage for each of those infill patterns, then warn the user if they exceed that maximum sensible infill percentage?

bubnikv, 2017-01-31:
That certainly makes sense. I believe the @alexrj slic3r implemented an automatic switch to a rectilinear pattern for high infill ratios recently.

4levels, 2017-09-08:
I noticed some over extrusions as well when using the new cubic infill at 50%: the nozzle started bumping the previous drawn lines more and more as infill layers where building up.  This sadly resulted in a slip of the y-axis belt causing the whole model to shift mid print.  I noticed this on the next attempt and had to reduce the speed from 40mm/s to 30 and increase the hotend temperature to prevent the issue from happening again.

rtyr, 2020-03-22:
User is warned in the current PrusaSlicer, I believe we can close this issue.
[image]

closed 2020-03-22 by rtyr