prusa3d/PrusaSlicer #68: "Cut" fz

tschaboo, 2016-12-19:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2016-12-20:
Would you please provide the STL, on which Slic3r always crashes? It certainly does not crash on most of the models, at least on my computer. Thanks.

bubnikv, 2017-01-20:
Did not receive any further information, closing.

closed 2017-01-20 by bubnikv